stm32duino/Arduino_Core_STM32 #219: STM32L151

karthigsn, 2018-02-13:
Not Able to flash stm32L151GBU6

[LoRa Module with STM32L151](http://www.rakwireless.com/en/module/lora/RAK811)

Thanks

fpistm, 2018-02-13:
Hi @karthigsn,
what did you expect exactly?
Currently, there is no variant to this module in the core.

karthigsn, 2018-02-14:
@fpistm 

http://www.rakwireless.com/en/module/lora/RAK811

Can you help me in programming the RAK811 which has STM32L151. I could not find Generic STM32L151 in Board manager of Arduino.

Thanks
Karthi

fpistm, 2018-02-14:
It is a module not a board like this one using STM32L052T8Y6 :
http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-3rd-party-evaluation-tools/i-nucleo-sx1272d.html
The RAK811 module comes with a preloaded AT-Command software and it can be controlled from an external host. You have to connect it to a board and send AT command to drive it.

closed 2018-02-14 by fpistm

sabas1080, 2018-02-14:
The RAK811 If it can be programmed, you used the IDE SW432 to do it. has the chip L151

fpistm, 2018-02-14:
There is several doc for this module:
http://www.rakwireless.com/en/download/RAK811/Software%20Development
Also the FW upgrade tools: