stm32duino/Arduino_Core_STM32 #220: Contribution: arduino-builder template file (and possible shortcoming)

aldrinleal, 2018-02-26:
Not an issue, but an contribution:

Here's a template for arduino-builder. However, a good point is that the current arduino-builder has an issue with the way that boards.txt is structured, thus needing to manually copy each and every option from boards.txt into -prefs

```
ARDUINO_PATH=$(HOME)/share/arduino-1.8.5
ARDUINO_USER_TOOLS=$(HOME)/.arduino15/packages/arduino/tools
ARDUINO_HARDWARE=$(ARDUINO_PATH)/hardware
ARDUINO_TOOLS=$(ARDUINO_PATH)/tools
ARDUINO_BUILDER=$(ARDUINO_PATH)/arduino-builder
ARDUINO_USER_LIBRARIES=$(HOME)/Arduino/libraries/
ARDUINO_CORE_LIBRARIES=$(ARDUINO_PATH)/libraries
STM32=$(HOME)/.arduino15/packages/STM32
STM32_HARDWARE=$(STM32)/hardware
STM32_TOOLS=$(STM32)/tools
RELEASE=2018.1.18
PLATFORM=stm32:$(RELEASE):Nucleo_64
STM32_MCU_FLAGS=-prefs=upload.maximum_size=1048576 -prefs=upload.maximum_data_size=131072 '-prefs=build.mcu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard' '-prefs=build.board=NUCLEO_L476RG' '-prefs=build.series=STM32L4xx' -prefs=build.product_line=STM32L476xx -prefs=build.variant=NUCLEO_L476RG -prefs=build.cmsis_lib_gcc=arm_cortexM4l_math -prefs=build.variant.path=$(STM32_HARDWARE)/stm32/$(RELEASE)/variants/NUCLEO_L476RG
ARDUINO_BUILDER_EXTRA_FLAGS=-verbose -debug-level 10
ARDUINO_BUILDER_FLAGS=$(ARDUINO_BUILDER_EXTRA_FLAGS) $(STM32_MCU_FLAGS) -fqbn $(PLATFORM) $(EXTRA_MCU_FLAGS) -hardware $(STM32_HARDWARE) -hardware $(ARDUINO_HARDWARE) -tools $(ARDUINO_USER_TOOLS) -tools $(STM32_TOOLS) -libraries $(ARDUINO_USER_LIBRARIES) -libraries $(ARDUINO_CORE_LIBRARIES) -prefs=runtime.tools.ctags.path=$(ARDUINO_PATH)/tools-builder/ctags/5.8-arduino11

%.elf: %.ino
	-mkdir $(PWD)/output/
	$(ARDUINO_BUILDER) $(ARDUINO_BUILDER_FLAGS) -build-path=$(PWD)/output -compile $(PWD)/$<

all: HelloWorld.elf
```

fpistm, 2018-03-06:
Hi  @aldrinleal 
Currently, I use a script based on Arduino CLI (which use arduino-builder in background).
But I'm considering to switch on arduino builder which I though is better for build.
I already test a script done by one of my colleague on Windows
https://github.com/fpistm/stm32_tools/tree/arduino-builder/arduino-builder/windows
I will write a new one in python or perl in order to be multi platform.

fpistm, 2018-07-31:
New script using arduino-builder released. 
See [CI/Build/](https://github.com/stm32duino/Arduino_Core_STM32/tree/master/CI/build)
So I close this one.
Feel free to comment

closed 2018-07-31 by fpistm