stm32duino/Arduino_Core_STM32 #223: Secure firmware update / firmware update within MCU

Adminius, 2018-03-07:
use case: device can not be connected to PCU via ST-Link or USB, because this device is build in "somewere" far away from PC (e.g. solar powered sensor at a tree with bluetooth connection to host).
at Embedded World 2018 i learned about [X-CUBE-SBSFU](http://www.st.com/en/embedded-software/x-cube-sbsfu.html)
My idea is to use SFU for such devices (transmit FW via bluetooth or other types of connections).
Question: is it possible to use this feature with this Arduino Core?

fpistm, 2018-03-08:
Hi @Adminius,
it could be possible but this will require to handle properly start address.
in fact, it's the same than handle properly a bootloader.

Adminius, 2018-03-08:
in my use case, i don't need security, i need only FW update from internal and/or external flash.
this library is for Arduino Zero.
[](https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SDU)
it makes it simmilar way, but with SD card. on the next days i will try to adopt it for [internal flash](https://github.com/cmaglie/FlashStorage). SPI=external flash is allready working :)

Normaly, user sketch will be placed on 0x2000 address (after bootloader)
SDU library has precompiled sketch (update routine) for MCU XYZ with a 0x4000 size. Then user writes own sketch and include sdu.h. that is. the linker script takes SDU-part an places it on 0x2000 address and user sketch on 0x6000. after MCU reset it starts SDU-part at 0x2000 that looking for update file and then switch to 0x6000 address => user sketch. so easy :D 

we can do it this way for STM32 MCUs, may be it is easier as SBSFU

fpistm, 2018-03-13:
Hi  @Adminius,
seems simple :)
Do not hesitate to provide a PR if you make a try +1:

Adminius, 2018-03-13:
i need example for FlashRead/Write (like FlashStorage) or a Library for that ;) 
and how i can change sketch beginn pointer...
than i can made a PR ;)

Adminius, 2018-03-13:
i will close this issue until we have FlashStorage library for STM32 MCUs. then we reopen it.

closed 2018-03-13 by Adminius

Adminius, 2018-03-21:
a little update:
i'm ready with Internal Flash Update (IFU) for SAMD21, that mean don't need any external HW to store update file. I do it on internal storage, but i loose about 50% of space.
FlashStorage like library for STM32 is not necessary, because HAL has already all functions, e.g. https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash.c