stm32duino/Arduino_Core_STM32 #227: NUCLEO-144 / NUCLEO-H743ZI (STM32H743ZI)

HackInvent, 2018-03-13:
Hi,
I'd like to know if there is a specific reason whether the NUCLEO-H743ZI isn't included.
We are willing to help to added it in the list of supported boards?
Thank you in advance,
Regards,
Alex

Adminius, 2018-03-13:
Reason: nobody made a PR yet ;)
Feel free to do it. :)

fpistm, 2018-03-14:
Hi @HackInvent 
There is no specific reason. H7 came later after the start of this core.
It will come.
First step will be to add HAL/LL, I've started this as a background task.
https://github.com/fpistm/Arduino_Core_STM32/tree/STM32H7
But this will required some work and tests.

HackInvent, 2018-03-14:
@fpistm  I didn't see your commit! that's a good news.
I will give a try to create a new variant for NUCLEO-H743ZI from your commit an try to blink the builtin led (GPIO).
If things go well with the GPIO I will move for the other peripherals.
To be continued

fpistm, 2018-03-14:
It's not a commit it's a working branch for H7 support.
Anyway, I will create the variant to test H7 support but before create it, it requires some work/test before.

HackInvent, 2018-03-15:
Ok, if there is a specific task/test we can do it on the board.

davidanger, 2018-05-11:
I have NUCLEO-F767ZI, can i help?

fpistm, 2018-05-11:
@davidanger this issue is for H7 not F7.
Anyway, an update about H7 support, I get the Nucleo H743 up and running with my branch. I've some issue to fix and it will be released

davidanger, 2018-08-23:
Now I got a NUCLEO-H743ZI, I'm working on this.

davidanger, 2018-08-23:
#308

fpistm, 2018-08-24:
unfortunately not, I'm OOO and this is on my local repo.
The first version is here as mentioned above:
https://github.com/fpistm/Arduino_Core_STM32/tree/STM32H7

mjs513, 2018-10-08:
Was wondering if there is any update on the timeline for incorporating the H7?  Otherwise I will be using fpistm version

fpistm, 2018-10-08:
Unfortunately, the next HAL/LL H7 is not released yet. I'm waiting it and I do not have any clue when this will be done.
I will try to get some info and then propose a first release.

mjs513, 2018-10-08:
Thanks appreciate it.  Looking at using it for some development efforts.

hasenbanck, 2018-10-16:
Just started a new project and wanted to use a H7 MCU. Too bad that the new HAL isn't ready yet. I will fall back to a F7 design for now.

fpistm, 2018-10-16:
Hi @hasenbanck 
I know but this is not in my hands. I think a way to release it as soon as possible but I don't think having it before end of the year.

hasenbanck, 2018-10-16:
@fpistm  Don't worry, I just wanted to show my interest. It's better to wait for the next HAL and use the new LL stuff / more stable HAL implementation.

fpistm, 2018-12-18:
Hi @HackInvent, @davidanger, @mjs513, @Adminius, @hasenbanck 
I've submitted the PR to support H7: #392 ;)

hasenbanck, 2018-12-18:
@fpistm It's seems you're pretty productive right now. Now you put me in a pinch and put pressure on me to change my current printer board project to use a H7...

HackInvent, 2018-12-23:
Merci Frederic!

HackInvent, 2018-12-29:
@fpistm I didn't dig around yet, but I was wondering what command are you using to upload the binary to the user flash, "stm32flash" tool?

fpistm, 2018-12-29:
Simply by copying binary on the mounted mass storage. 
Or you can use STM32CubeProgrammer install on your host machine and selecting the binary generated by Arduino.

PR #392 (2019-01-04): Add STM32 H7 series support and Nucleo H743ZI variant
This PR add the STM32 H7 series support based on STM32 H7 HAL and CMSIS v 1.4.0.
This also add the support of [NUCLEO-H743ZI](https://www.st.com/en/evaluation-tools/nucleo-h743zi.html)

# Restrictions: 
 * Upload only available thanks the Mass Storage. STLink is not updated to support H7 target, anyway it is possible to use the STM32CubeProgrammer to upload manually the binary.
 * Ethernet currently not supported.

## Test performed:
 * GPIO
 * ADC
 * DAC
 * PWM
 * I2C (EEPROM RW, I2C scanner)
 * SPI (SD, TFT)
 * EEPROM emulation (put/get o [... cut]
closed 2019-01-04 by fpistm