stm32duino/Arduino_Core_STM32 #228: Disable HAL  feature when not used in the sketch

fpistm, 2018-03-16:
Goal of this issue is to be able to disable HAL module if not used in the sketch.
ex: I2C, SPI, ADC, PWM,...

fpistm, 2018-04-09:
This will be available for all variants. It's a core feature not specific to a variant.
Anyway, depending of the feature used, the HAL part is not embed if not used.
Example for SPI or I2C if SPI.h or Wire.h is not include, the HAL part is removed by the compiler as it is not used. Only the pinmap array is kept but it should be removed if not used to save some space.

fpistm, 2018-06-15:
Serial can now be enabled or not:
https://github.com/stm32duino/Arduino_Core_STM32/commit/83d872ff42c3d4988ffed7295199b22a453c9362

This allow to build without HAL_UART_MODULE_ENABLED (save memory)
HAL_UART_MODULE_ENABLED should not be defined in stm32yyxx_hal_conf.h

tve, 2018-08-05:
Shouldn't RCC also be on the list here, e.g., following the comment https://github.com/stm32duino/Arduino_Core_STM32/issues/165#issuecomment-373646769 ("I've generate the code with CubeMx to use LL instead of HAL for RCC") ?

fpistm, 2018-08-06:
This could.

Adminius, 2018-11-02:
hi @fpistm ,
for this points:

> ADC: be able to build without HAL_ADC_MODULE_ENABLED
>  DAC: it's already possible to build without HAL_DAC_MODULE_ENABLED, maybe a menu could be proposed to enable or not the feature...
>  PWM: be able to build without HAL_PWM_MODULE_ENABLED

we need menu and some ```ifdef``` "only", right?

fpistm, 2018-11-02:
Hi @Adminius,
probably yes, I have to find a nice and easiest way to do this.

closed 2019-05-15 by fpistm