stm32duino/Arduino_Core_STM32 #239: error when installing STM32 core support for Arduino

ED189, 2018-05-01:
I am getting simple error when I try to install on Arduino IDE (the latest), it says the zip file does not contain library

per1234, 2018-05-01:
As it should since this is not a library.

If you follow the instructions you would not have this problem:
https://github.com/stm32duino/wiki/wiki/Getting-Started

fpistm, 2018-05-02:
Hi,
as mentioned by @per1234 , please follow the getting started wiki page. (thanks @per1234 )

ED189, 2018-05-02:
when you work with arduino IDE the is menu option sketch ->Include libray -> add library from zip file

it does not recognize it as a zipped library file, there is no other way to add library it has to be in zip file

ED189, 2018-05-02:
the link you gave me is to add board support in Arduino I did it and it worked with no problems

I am talking about adding library not the board

ED189, 2018-05-02:
this is the library

https://github.com/stm32duino/Arduino_Core_STM32

I am trying to add