stm32duino/Arduino_Core_STM32 #240: va_start & va_end are not defined

Adminius, 2018-05-04:
i'm missing this two functions:

- `va_start`
- `va_end`

i had success with this workaround:

> `if defined(ARDUINO_ARCH_STM32)
> #include <stdarg.h>
> #define va_start(v,l)	__builtin_va_start(v,l)
> #define va_end(v)	__builtin_va_end(v)
> #endif

in which file i can make PR? :)

fpistm, 2018-05-05:
I think you could simply include stdio.h here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/bad5b56870ce311a4d591f789cbe5d47450097e4/cores/arduino/wiring.h#L23

Adminius, 2018-05-07:
if add my implementaion in wiring_privat.h it doesn't work. 
But it works in wiring.h...

P.S. im on vacation in paris until weekend, so i'm not available...

fpistm, 2018-05-17:
Hi @Adminius ,
could you share with me your use case to understand why it's not working with wiring_private.h

Adminius, 2018-05-17:
we use in our [KONNEKTING Library](https://github.com/KONNEKTING/KonnektingDeviceLibrary/tree/develop_beta4a) a Debug Class, that uses [this to functions](https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/develop_beta4a/src/DebugUtil.cpp#L59).

You can complie a [DemoSketch ](https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/develop_beta4a/examples/DemoSketch/DemoSketch.ino)with [KDEBUG enabled](https://github.com/KONNEKTING/KonnektingDeviceLibrary/blob/develop_beta4a/examples/DemoSketch/DemoSketch.ino#L9) to test it

fpistm, 2018-05-19:
@Adminius 
I've tested and build is now ok.
Let me know if it is ok then I will merge #253

Adminius, 2018-05-20:
Hi @fpistm 
it works, thx. 
I will accept your PR after new STM32 core will be released, deal?

Issue can be closed.

fpistm, 2018-05-20:
Of course ;)
New release is planned this week or the next one.

commit 041c616e (2018-05-21): Update wiring_private.h to include stdarg.h

va_start & va_end was not defined
To have them defined include wiring_private.h

Fix #240