stm32duino/Arduino_Core_STM32 #243: ERROR in reading ADS1220 ADC through SPI communication

Vicky-S, 2018-05-10:
Please, before reporting any issue:
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported or discussed on the [stm32duino forum](http://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the Arduino_Core_STM32.
- **Avoid** to submit a GitHub issue for project troubleshooting.
 
Any questions/feedback/sugestions should be discussed on the [stm32duino forum](http://stm32duino.com):
  * questions on the [STM32 Core](http://stm32duino.com/viewforum.php?f=48)
  * bugs/enhancements on the [STM core: Bugs and enhancements](http://stm32duino.com/viewforum.php?f=49)

When reporting any issue, please try to provide all relevant information:

- Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)
- Which STM32 based board you used
- If other hardware is required to reproduce the issue, please provide links or part numbers where this hardware can be purchased.
- Which version of STM32 core and Arduino, Tools menu settings, operating system, ...

Thanks

fpistm, 2018-05-10:
You probably declare a Serial instance with pins which are not on the same USART peripheral:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/uart.c#L117
Without your sketch and which board used, It's difficult to help you.

Vicky-S, 2018-05-11:
Im using STM32F746NG board. ADC i was used is ADS1220 which uses SPI communication for getting ADC reading. And used default Serial UART for checking the received data in serial monitor.

Its a open source code. I shared the link below for your Reference.

https://github.com/Protocentral/Protocentral_ADS1220

Vicky-S, 2018-05-11:
I kept everything in single file( including h file and cpp file). Removed the class function and made everything as local functions in single file. And its working fine now. But i didnt changed any pins in the program and its not showing any error. I dont know how its fixed. Im trying to find the reason.

Vicky-S, 2018-05-11:
"
  pinMode(ADS1220_CS_PIN, OUTPUT);
  pinMode(ADS1220_DRDY_PIN, INPUT);

  ADS1220.begin();
  Serial.begin(9600);
  //ADS1220.Protocentral_ADS1220();   
  Serial.println("Welcome");
"
I made the changes as per your instructions. Now also it gives the same error. but however it works fine after i kept everythting in single file. 

Thanks.

fpistm, 2018-05-11:
I use the defaut example without any issue after commenting Serial in the constructor.
You should not call `Serial.begin(9600);` this is already done in the `ADS1220.begin();`
What do you mean by ?

> i kept everythting in single file

Vicky-S, 2018-05-11:
ok my understanding was wrong. It works fine after commenting serial in constructor.