stm32duino/Arduino_Core_STM32 #244: Where can I find PINMAP for multiple serial?

davidanger, 2018-05-11:
I'm using NUCLEO-F411RE.
How can i using multiple serial?
What's different between Serial Interface select "Generic Serial", "All (up to 10)" and "Enable first third (USART 1 to 3 if available)?

Thanks!

fpistm, 2018-05-11:
This part will be reviewed as it is not user friendly and has no real interest.
Generic Serial instantiate the Serialx instance linked to U(S)ARTx mainly connected to STLink to benefits the VCP com port. This Serialx instance is aliased to Arduino standard Serial name.
Anyway to use one other U(S)ART simply declare it:
`HardwareSerial mySerial(rx,tx)`

For F411, Serial is Serial2 (using USART2).
Here the list of all available U(S)ART:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/NUCLEO_F411RE/PeripheralPins.c#L143

So to use for example USART1:
`HardwareSerial mySerial(PA10,PA9);`
or
`HardwareSerial mySerial(USART1);`

davidanger, 2018-05-17:
So the NUCLEO-F411RE Only 3 Hardware Serial can be used.
By default:
Serial = Serial2 USART2 TX:D1(PA2) RX:D0(PA3) STLink Virtual COM
Serial1 USART1 TX:D8(PA9) RX:D2(PA10) 
Serial6 USART6 TX:PA11 RX:PA12

fpistm, 2018-05-17:
Right,
USART6 could be used also on PC7/PC6 or a mix Rx(PA11 or PC7)  and Tx (PA12 or PC6)
Same for USART1 which have 3 pins possibilities per Signal.

fpistm, 2018-05-19:
Hi, @davidanger is it ok to close this?

davidanger, 2018-05-19:
OK. And does this project have wiki?

closed 2018-05-19 by fpistm