stm32duino/Arduino_Core_STM32 #247: STM32F7 doesnt have LCD/TFT and touch support.

Vicky-S, 2018-05-12:
I tried to explore the LCD/TFT interface after checking the SD card, Serial, GPIO, EEPROM, SPI communication. Everything is working fine except LCD/TFT interface. I tried to use the LTDC interface from STM32GENERIC for STM32 official core but its not working.

Is there any example for LCD of STM32F7 in official core? Please share it if there is any.

Thank you.

fpistm, 2018-05-12:
Here a quick port of @danieleff and @ChrisMicro to test LTDC and TS:
https://github.com/fpistm/Arduino_Core_STM32/tree/LTDC_TS_Test
This will not be merge, it only for test purpose.
A generic LTDC and TS will need to be study.

Vicky-S, 2018-05-14:
Thanks bro. I have checked the LTDC and TS example its working. I'll test more and give feedback to u.

Vicky-S, 2018-05-14:
It works very fine. I have tested with gui library from STM32GENERIC because with the help of gui only we can create windows, text boxes, call back functions etc. Since u made some changes in touch library i need to do some modifications on touch part of the program. After some modifications it works very fine. 

But when we continuously updating the screen it shows some glitch on the screen.

fpistm, 2018-05-14:
The only changes for TS is:
```
-    Wire.stm32SetInstance(I2C3);
-    Wire.stm32SetSDA(PH8);
-    Wire.stm32SetSCL(PH7);
+    Wire.setSDA(PH8);
+    Wire.setSCL(PH7);
```
And that I put the TS as a library. @ChrisMicro update his TS example after @danieleff merge it:
https://github.com/ChrisMicro/Arduino_STM32F7_Discovery/tree/master/_0005_TouchDisplayTemplate
For glitch I've disabled the D-Cache in all the example. Don't know if you disable it for your sketch.

Vicky-S, 2018-05-14:
They kept TS as local library and initialized in every example. In official core u are inserted the library in git repo itself so we dont need to add it in every example which is good for the users.

They got the coordinates from the local function itself. In official core we need to create a object to call the function "getPoint()" and the variables of the co ordinates also changed.

These are the minor changes but if we load the generic program directly after commenting the I2C instance in official core, it wont work.

Thanks for making the TS as inbuilt library

Vicky-S, 2018-05-14:
glitch is still there after i added "build_opt.h". do we need to add this library

fpistm, 2018-05-14:
> glitch is still there after i added "build_opt.h". do we need to add this library
build_opt.h is not a library. This is a way to add some build option at sketch level (undef, define,  add  include path, static lib,...) for gcc

Vicky-S, 2018-05-14:
Oh.. thanks... I didnt included that library... I just copied that build_opt.h from one of the example and place it on my sketch folder... But glitch is still there when we continuous update the textbox...

closed 2018-05-18 by Vicky-S

sixtyfive, 2020-06-12:
This issue was closed, but I don't see anything in the [libraries folder](https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries) about either LCD/TFT or the capacitive touch controller peripheral. Would it be possible to re-open the issue? Or can your `LTDC_TS_Test` branch already be merged, @fpistm?

fpistm, 2020-06-12:
Hi @sixtyfive 

As mentioned above:

> This will not be merged, it is only for test purpose.
> A generic LTDC and TS will need to be study.

sixtyfive, 2020-06-12:
So the issue is still open: it has not been studied and the feature is not implemented, correct?

fpistm, 2020-06-12:
> So the issue is still open: it has not been studied and the feature is not implemented, correct?

Time is always the missing ingredients. 😬  Any contributions are welcome. The OP closed this issue as he got a way to use it.

sixtyfive, 2020-06-12:
Okay, fair enough! :) I'd love to help, unfortunately I'm only a beginner with an Arduino-beginner-level understanding of C++. Hopefully someone will come along!