stm32duino/Arduino_Core_STM32 #252: How can i load images on STM32F7 TFT?

Vicky-S, 2018-05-19:
How can i load images on STM32F7. There is an example for loading images and its working fine but how can i load my own image on display.?

fpistm, 2018-05-19:
You should have a look to STM32Cube F7 example.
Display/LTDC_PicturesFromSDCard/

Vicky-S, 2018-05-19:
Ok sorry.

closed 2018-05-19 by Vicky-S

Vicky-S, 2018-05-19:
But there is still an issue in STM32F7 EEPROM. I dont have any other STM board except F746 disco to test the EEPROM but it takes more than 2 secs to write single byte data.  I have 3 nos. STM32F7 i checked in all the 3 boards to check whether its an hardware problem but it not. Can u please check it once?

fpistm, 2018-05-19:
Nous worries😉 I will check eeprom emulation when I will got Time. Time is Always the missing ingredient 😅

Vicky-S, 2018-05-19:
Ok.. thanks bro...

Adminius, 2018-05-20:
fyi: i have no issues with EEPROM on F401 and L476

Vicky-S, 2018-05-21:
Hi... If u have STM32F7 discovery board can u please check it.. So we can sure about whether its an common issue for STM32F7  or not?

fpistm, 2018-05-21:
It should be better to talk about the issue on its one :) --> #248