stm32duino/Arduino_Core_STM32 #258: Arduino_Core_STM32 1.3.0 used for STM32lowpower when and where to download

jackyouandme, 2018-05-31:
Please, before reporting any issue:
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported or discussed on the [stm32duino forum](http://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the Arduino_Core_STM32.
- **Avoid** to submit a GitHub issue for project troubleshooting.
 
Any questions/feedback/sugestions should be discussed on the [stm32duino forum](http://stm32duino.com):
  * questions on the [STM32 Core](http://stm32duino.com/viewforum.php?f=48)
  * bugs/enhancements on the [STM core: Bugs and enhancements](http://stm32duino.com/viewforum.php?f=49)

When reporting any issue, please try to provide all relevant information:

- Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)
- Which STM32 based board you used
- If other hardware is required to reproduce the issue, please provide links or part numbers where this hardware can be purchased.
- Which version of STM32 core and Arduino, Tools menu settings, operating system, ...

Thanks

fpistm, 2018-05-31:
Hi @jackyouandme 
The 1.3.0 will be release next week. I'm currently working on this.
STM32RTC and STM32LowPower will be also release and I will ask to Arduino to reference them in the Library manager.

dds90, 2018-06-10:
Hi Frederic!
I need to use STM32RTC lib. When will be release the 1.3.0? Meantime, How can I use a preview?
Thank you in advance!

fpistm, 2018-06-10:
Hi @dds90.
I've got some issue that's why I've delayed it. 
I hope release it end of this week.
To use the git:
https://github.com/stm32duino/wiki/wiki/Using-git-repository
and use the zip of the STM32RTC library

fpistm, 2018-06-15:
Release done and available thanks Arduino Board manager:
https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.3.0

Libraries released 
[STM32RTC 1.0.0 ](https://github.com/stm32duino/STM32RTC/releases/tag/1.0.0)
[ STM32LowPower 1.0.0](https://github.com/stm32duino/STM32LowPower/releases/tag/1.0.0)

and available thanks Arduino library manager soon:
https://github.com/arduino/Arduino/issues/7696

dds90, 2018-06-15:
Hi @fpistm,
Thank you so much for your support!
I confirm that work well with RTC_LSE_CLOCK in a STM32VLDISCOVERY.
To keep the time with the external backup battery, I must use also the LowPower library?
Thank you again,
Daniele

fpistm, 2018-06-16:
Welcome.
Not necessarily, but this could help to save power.

fpistm, 2018-06-16:
I close this issue has 1.3.0 and libs are released.

closed 2018-06-16 by fpistm

jackyouandme, 2018-06-16:
Hi  Frederic Pillon,

   Thank you very much for your hard work.

terry

jackyouandme@foxmail.com
 
From: Frederic Pillon
Date: 2018-06-16 13:17
To: stm32duino/Arduino_Core_STM32
CC: jackyouandme; Mention
Subject: Re: [stm32duino/Arduino_Core_STM32] Arduino_Core_STM32 1.3.0 used for STM32lowpower when and where to download (#258)
Closed #258.