stm32duino/Arduino_Core_STM32 #259: CAN communication support not available for STM32F746

Vicky-S, 2018-06-02:
I could not find any example for CAN communication in STM32 official core. Is there any?

fpistm, 2018-06-02:
Currently, there is no CAN support

Vicky-S, 2018-06-02:
Ok. will it be available in future?

fpistm, 2018-06-02:
Yes but I can't tell when...

Vicky-S, 2018-06-02:
Ok... Please try to make support ASAP..

BennehBoy, 2019-02-11:
@fpistm, by no support you mean there is no library right?  So people could program directly against HAL  CAN modules?

fpistm, 2019-02-11:
@BennehBoy 
Yes of course. All STM32 features are available thanks HAL/LL (if user enable the right HAL module).
All stuff which can be done thanks the cube are available. By no support I told about Arduino point of view, there is no dedicated library to support STM32 CAN for this core with specific Arduino API style

jacky4566, 2019-11-11:
Bump. Would love to see some CAN bus support. Seems silly to buy a standalone SPI chip when all you need is a transceiver.

fpistm, 2019-11-12:
Hi @jacky4566 
I'd love too. 
I know @geosmall made some work around CAN:
https://github.com/geosmall/UAVCAN-for-STM32-Arduino

seeers, 2020-01-08:
Unfortunately the UAVCAN example only works with the STM32Duino version 1.4.
With version 1.8 I get the following errors:
`
sketch\can.c: In function 'MX_CAN_Init':
can.c:12: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'SJW'
   12 |   hcan.Init.SJW = CAN_SJW_1TQ;
      |            ^
can.c:13: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'BS1'
   13 |   hcan.Init.BS1 = CAN_BS1_1TQ;
      |            ^
can.c:14: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'BS2'
   14 |   hcan.Init.BS2 = CAN_BS2_1TQ;
      |            ^
can.c:15: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'TTCM'
   15 |   hcan.Init.TTCM = DISABLE;
      |            ^
can.c:16: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'ABOM'
   16 |   hcan.Init.ABOM = DISABLE;
      |            ^
can.c:17: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'AWUM'
   17 |   hcan.Init.AWUM = DISABLE;
      |            ^
can.c:18: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'NART'
   18 |   hcan.Init.NART = DISABLE;
      |            ^
can.c:19: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'RFLM'
   19 |   hcan.Init.RFLM = DISABLE;
      |            ^
can.c:20: error: 'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'TXFP'
   20 |   hcan.Init.TXFP = DISABLE;
      |            ^
exit status 1
'CAN_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'SJW'

`

Is there maybe an easy way to get it running with the current version?

fpistm, 2020-01-08:
@seeers 
I think this is due to STM32F1 HAL update.
looking at the Release_Notes.html:
https://htmlpreview.github.io/?https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32F1xx_HAL_Driver/Release_Notes.html

From version V1.1.3 / 09-October-2018:

> HAL CAN update
> Fields of CAN_InitTypeDef structure are reworked:
> SJW to SyncJumpWidth, BS1 to TimeSeg1, BS2 to TimeSeg2, TTCM to TimeTriggeredMode, ABOM to AutoBusOff, AWUM to AutoWakeUp, NART to AutoRetransmission (inversed), RFLM to ReceiveFifoLocked and TXFP to TransmitFifoPriority
> HAL_CAN_Init() is split into both HAL_CAN_Init() and HAL_CAN_Start() API’s
> HAL_CAN_Transmit() is replaced by HAL_CAN_AddTxMessage() to place Tx Request, then HAL_CAN_GetTxMailboxesFreeLevel() for polling until completion.
> HAL_CAN_Transmit_IT() is replaced by HAL_CAN_ActivateNotification() to enable transmit IT, then HAL_CAN_AddTxMessage() for place Tx request.
> HAL_CAN_Receive() is replaced by HAL_CAN_GetRxFifoFillLevel() for polling until reception, then HAL_CAN_GetRxMessage()
> to get Rx message.
> HAL_CAN_Receive_IT() is replaced by HAL_CAN_ActivateNotification() to enable receive IT, then HAL_CAN_GetRxMessage()
> in the receivecallback to get Rx message
> HAL_CAN_Slepp() is renamed as HAL_CAN_RequestSleep()
> HAL_CAN_TxCpltCallback() is split into HAL_CAN_TxMailbox0CompleteCallback(), HAL_CAN_TxMailbox1CompleteCallback() and HAL_CAN_TxMailbox2CompleteCallback().
> HAL_CAN_RxCpltCallback is split into HAL_CAN_RxFifo0MsgPendingCallback() and HAL_CAN_RxFifo1MsgPendingCallback().
> More complete “How to use the new driver” is detailed in the driver header section itself.

It seems there is an `HAL_CAN_LEGACY_MODULE_ENABLED` which can be enabled to use the legacy one but `HAL_CAN_MODULE_ENABLED` have to be disabled.

jacky4566, 2020-01-08:
If you look at the code in this project the Canbus registers are really not all that hard to use. Maybe 3 or 4 key regs. 
I got it working after working out some of the defined values. I am working on my own implementation for STM32L4 as part of a custom library. 

https://github.com/UBC-Solar/Firmware-v2/tree/master/Peripherals/CAN

geosmall, 2020-01-10:
I have the Libcanard UAVCAN example mentioned above now up and running / verified on V1.8.
https://github.com/geosmall/UAVCAN-for-STM32-Arduino

As this library (https://github.com/UAVCAN/libuavcan) uses it's own low level access and does not use HAL/LL I only needed to fix the rename deltas so not too bad.

UAVCAN based CAN is gaining some traction in both Ardupilot (https://github.com/ArduPilot/ardupilot) and PX4 (https://github.com/PX4/Firmware) so I'm planning to stay on that path.

I will update my example project over the weekend and document on https://www.stm32duino.com/

fpistm, 2020-01-10:
Thanks  @geosmall for the feedback.

You can also document on the wiki if you want.

Vicky-S, 2020-01-10:
Will it work for STM32F4 Series too?

seeers, 2020-01-10:
> If you look at the code in this project the Canbus registers are really not all that hard to use. Maybe 3 or 4 key regs.
> I got it working after working out some of the defined values. I am working [...]

I got it running with parts of your code on an stm32f103c8t6. Thank you very much

geosmall, 2020-01-11:
@Vicky-S I have exampl

> Will it work for STM32F4 Series too? @geosmall

I have examples for F103C8 (Bluepill) AND F303RE-NUCLEO I am updating.  They are all pretty close but the devil is in the datasheets :).  I can check on my NUCLEO-F411RE.

geosmall, 2020-01-11:
> Will it work for STM32F4 Series too? @geosmall

> Thanks @geosmall for the feedback.
> 
> You can also document on the wiki if you want.

Ok I'll capture on the forum and then figure out how to contribute to wiki (another first for me so forgive my mistakes in advance I will fix as I go :)

geosmall, 2020-01-13:
UAVCAN libcanard based CAN example updated for v1.8 here https://github.com/geosmall/UAVCAN-for-STM32-Arduino.

F1 preliminary coms testing completed so far.

Vicky-S, 2020-01-14:
@geosmall 

> The CAN periperals are all pretty close but the devil is in the datasheets :).

> It would of course need an F4 variant with CAN peripheral (e.g. F405/7, F446, etc and not F401, F411...).

I have F446 Board. Please tell me about the required changes. Let me try.

geosmall, 2020-01-15:
F103 Bluepill end-to-end CAN testing now complete.  ESC values input with UAVCAN GUI are sent over CAN bus to Bluepill node, which receives and re-transmits those same ESC values out over Serial.

I suggest further discussion move to the forum here.: https://www.stm32duino.com/viewtopic.php?f=10&t=99

geosmall, 2020-01-19:
F103C8 and F303CC UAVCAN functional testing now completed for both examples.

pazi88, 2020-03-03:
Is there any progress on CAN support for STM32? It would be really useful for many projects. The example code from @seeers for example would be enough for me currently, but I can seem to figure out how to use the PA11 and PA12 in stm32f103c8t6 instead of the PB8 and PB9 that the code does remapping of the pins. https://github.com/seeers/CAN-Bus-Arduino_Core_STM32

fpistm, 2020-03-04:
@pazi88 
currently, there is no dedicated task for this on my side.

orhaneee, 2020-03-16:
@pazi88 how did you edit the code in order to use PA11 and PA12?

pazi88, 2020-03-16:
@orhaneee See the code in here: https://github.com/pazi88/8Ch-EGT I have included CAN code to mine, but you should be able to make that out. The register settings are slightly different depending what pins you want to use. That all is verified to work.

nopnop2002, 2020-03-25:
@orhaneee 
How to remap CAN port for stm32f103:
https://github.com/nopnop2002/Arduino-STM32-CAN/tree/master/stm32f103

pazi88, 2020-03-25:
@nopnop2002 Thanks for that! I have tried to figure out how to make CAN work on F4 too, but that will save lot of time.

nopnop2002, 2020-03-25:
@pazi88 I am currently trying to send and receive complete interrupts but have not yet succeeded.

Elektrik1, 2020-09-26:
Any progress on implementing Hardware CAN library?

J-f-Jensen, 2020-10-08:
I have made a working CAN library based on libraries I have found. It should work both with can0 and can1 on devices with two CAN devices, but I have only tested it with CAN0. USB serial cannot be used together with the CAN library on stm32f103 bluepill devices. Link: https://github.com/J-f-Jensen/libraries

fpistm, 2020-10-08:
Thanks for sharing @J-f-Jensen 
Maybe it could be a good candidate to add it as a built-in library or other.
Probably some update would be required to support all series but I guess 90% of the work is done. 👏

lutfisan, 2020-11-10:
any CAN FD support for NUCLEO-G474RE?

pazi88, 2020-11-26:
> I have made a working CAN library based on libraries I have found. It should work both with can0 and can1 on devices with two CAN devices, but I have only tested it with CAN0. USB serial cannot be [...]

Nice library, but what I can see it only works on F1. It fails to build on F4.

J-f-Jensen, 2020-11-26:
What error do you get? I can be a issue with the hal config fileBr Jens 
-------- Oprindelig besked --------Fra: Pasi Kemppainen <notifications@github.com> Dato: 26.11.2020  14.13  (GMT+01:00) Til: stm32duino/Arduino_Core_STM32 <Arduino_Core_STM32@noreply.github.com> Cc: J-f-Jensen <jens@2stroke4fun.dk>, Mention <mention@noreply.github.com> Emne: Re: [stm32duino/Arduino_Core_STM32] CAN communication support (#259) 

I have made a working CAN library based on libraries I have found. It should work both with can0 and can1 on devices with two CAN devices, but I have only tested it with CAN0. USB serial cannot be used together with the CAN library on stm32f103 bluepill devices. Link: https://github.com/J-f-Jensen/libraries

Nice library, but what I can see it only works on F1. It fails to build on F4.

pazi88, 2020-11-30:
For example there is no AFIO registers in F4. And there was some other errors too.

fpistm, 2020-11-30:
Hi @pazi88 
Which F4 boards you used ? did you defined the `HAL_CAN_MODULE_ENABLED `?

https://github.com/stm32duino/wiki/wiki/HAL-configuration#hal-modules-configuration

shaneyake, 2020-12-03:
Does anybody know of a library that will work with the HAL_FDCAN_MODULE_ENABLED?
@J-f-Jensen library is great but I can't get it to compile for my Nucleo-H743IZ2.
or is there a way to use HAL_CAN_MODULE_ENABLED with the Nucleo-H743IZ2?
Thanks

pazi88, 2020-12-03:
@shaneyake The library only works for F1 series as far as I know, because that's the only one using AFIO for pin mapping. F4 andvothets have different way to map pins, so that's why it wont compile (as I mentioned above).

seb43654, 2022-05-08:
any progress on this? would be very useful to have can support...

pazi88, 2022-05-08:
> any progress on this? would be very useful to have can support...

You can use this library I created. Uses HAL drivers, hadware interrupts and it's easy yo use. Should work with every stm32 chip that is supported by stm32duino: https://github.com/pazi88/STM32_CAN

pazi88, 2022-05-08:
I have been meaning to do some last modifications to that and maybe make a PR for that to be added in stm32core.

seb43654, 2022-05-08:
Wow thanks @pazi88  will test tomorrow! Was previously using MBed but it is failing with high bus loads....

fpistm, 2022-05-09:
> e last modifications to that and maybe make a PR for that to be added in stm32core. I have been personally used that with many projects communicating busy vehicle CAan busses and it works great (other ones I have found, have many problems).

Hi @pazi88, do not hesitate to male a PR or you can also share it as an Arduino library.

pazi88, 2022-05-09:
> Wow thanks @pazi88 will test tomorrow! Was previously using MBed but it is failing with high bus loads....

Yes with other libraries available I also had problems with high bus loads, but with this one all seems fine, because both TX and RX side are interrupt based and the packets are gathered to buffers that have configurable size.

pazi88, 2022-05-09:
> Hi @pazi88, do not hesitate to male a PR or you can also share it as an Arduino library.

It works already as Arduino library. But I'll do those additions I have been meaning to do and do a PR then.

ChinaBurn, 2022-05-25:
There is a working Can instance here, but this is not a library, I have tested it to work and use it in production.
https://github.com/nopnop2002/Arduino-STM32-CAN