stm32duino/Arduino_Core_STM32 #260: UART7 causes Nucleo F091RC to lock up

TT-ZX, 2018-06-02:
When I try to use Serial.print over UART7 on the Nucleo F091RC the MCU locks up.
Here is the code to demonstrate the issue:

`HardwareSerial mySerial(PC7,PC6);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  mySerial.begin(9600);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  mySerial.println("Hello World!");
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}`

The LED will turn on and stay on.  Nothing is received from UART7
I am using V1.2.0 of STM32 Cores.  I've tried earlier version but the result is the same.  I am using version 1.8.5 of the Arduino IDE on Gentoo.

Can anyone else confirm this issue or know of a work around?

Thanks,
Justin

TT-ZX, 2018-06-02:
Is that something I can easily change and try or is it quite involved?

TT-ZX, 2018-06-02:
I tried UART3 - UART8 and got the same problem.  Only UART1 and UART2 work for me.

fpistm, 2018-06-02:
F091 is a bit specific for UART.
I think issue is probably linked to uart irq handler.
https://github.com/stm32duino/Arduino_Core_STM32/blob/22290391347075c19152716efe0c95bda5e9c6ac/cores/arduino/stm32/uart.c#L779

Maybe you can try comment here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/22290391347075c19152716efe0c95bda5e9c6ac/cores/arduino/stm32/uart.c#L796

this test:
`if (__HAL_GET_PENDING_IT(HAL_ITLINE_USART7)!= RESET)`

TT-ZX, 2018-06-03:
This works for all but UART6 and only works for one UART at a time.  If I comment more that one of those tests, only the last UART works.

fpistm, 2018-06-03:
Ok thanks for the test. I will test and try to fix it for the next release.

fpistm, 2018-06-14:
Hi, @TT-ZX 
I've found the issue.
__HAL_GET_PENDING_IT() read IT line status in SYSCFG register which required to be clocked.
    
This set of registers is implemented only on STM32F09x  to collect all pending interrupt sources associated with  each interrupt line into a single register.
This allows users to check by single read which peripheral requires service in case more than one source is associated to the interrupt line.

I've tested also with several USART in the same sketch:
```
HardwareSerial Serial7(PC7,PC6);
HardwareSerial Serial4(PA1,PA0);
HardwareSerial Serial3(PB_11,PB_10);

void setup() {
Serial3.begin(115200);
Serial4.begin(115200);
Serial7.begin(9600);
}

void loop() {
Serial3.println("Hello World I'm USART3!");
Serial4.println("Hello World I'm USART4!");
Serial7.println("Hello World I'm USART7!");
}
```

PR #264 (2018-06-15): Serial update
This PR includes several update around Serial feature;

* Review Serial menu, 3 options:
      - Enabled with generic Serial: default, define HAL_UART_MODULE_ENABLED
        and instantiate Serial using SERIAL_UART_INSTANCE and PIN_SERIAL_RX/PIN_SERIAL_Tx
      - Enabled without generic Serial: define HAL_UART_MODULE_ENABLED
      - Disabled (No Serial): HAL_UART_MODULE_ENABLED not defined
    
  Allow to build without HAL_UART_MODULE_ENABLED (save memory)
  Now, HAL_UART_MODULE_ENABLED should not be defined in stm32yyxx_hal_conf.h
* Add LPUART1 management
* Fix serial_[rx|tx]_ac [... cut]
commit 3adfa857 (2018-06-15): [STM32F09x] Fix Serial using USART[3-8]

__HAL_GET_PENDING_IT() read IT line status in
SYSCFG register which required to be clocked.

This set of registers is implemented only on STM32F09x
to collect all pending interrupt sources associated with
each interrupt line into a single register.
This allows users to check by single read which peripheral
requires service in case more than one source is associated
to the interrupt line.

Fix #260
closed 2018-06-15 by fpistm