stm32duino/Arduino_Core_STM32 #261: ADC and PWM Problem with stm32f101vb

atc1441, 2018-06-05:
Hi.
I use the newest version of the Stm32 Core 

I added a new board as descrited in the manual
Because i own 10 Vacuum cleaner with the STM32F101VB on board.

The main things are working like serial, digitalwrite/read and sleep.

But the  timer and analog write is making some trouble, when i use a bluepill analog write and read works fine.

I will post my config files tomorow 

maybe someone has a clue what could be wrong. 

Also i do wonder why is there no open drain option in the core ? I wrote my own method to use it maybe this is something for everybody.
Its quite simple to add.

Thanks for the core :)

fpistm, 2018-06-06:
Hi,

maybe the trouble comes from clock config about PWM.
About Open drain, Yes this could/sjould be added. Why there is not actually, is because the first port was Arduino Uno like and there is no open drain.

atc1441, 2018-06-06:
ok i got it somehow to work 

i change these value 

#define NUM_ANALOG_INPUTS       16
#define NUM_ANALOG_FIRST        16

from 10 to 16 and it works but,

 the pins are not if read like this: `analogRead(PA1)`

it reads from: PB1

fpistm, 2018-06-06:
It seems you do not defined properly ANALOG pins.
In enum in variant.h:
PA1 is 1
So analogRead(PA1) is equivalent to analogRead(1)
As you set NUM_ANALOG_FIRST 16
This result to read on 1+16
PB1 = 17

Analog pins must be contiguous to be able to loop on each value
And first analog pin value (NUM_ANALOG_FIRST) must be greater than or equal to the number of analog (NUM_ANALOG_INPUTS)
This assert is here to avoid this issue
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/pins_arduino.h#L55
but as you define wrong NUM_ANALOG_FIRST and NUM_ANALOG_INPUTS

atc1441, 2018-06-06:
Thank you for that, that was correct,  

i arranged the analog inputs after the datasheet order and made the #define NUM_ANALOG_INPUTS 16 and #define NUM_ANALOG_FIRST 16
to the right position.

Now i will take deeper look in to analogwrite.

closed 2018-06-06 by atc1441