stm32duino/Arduino_Core_STM32 #272: Version 1.3.0 the Serial function is not compatible with V1.2.0

davidanger, 2018-06-28:
In 1.2.0 I using the config "Serial interface: All(up to 10)"
In code I just using 
Serial1.begin(); 
Serial6.begin();
like this.
But in 1.3.0, This not worked.

fpistm, 2018-06-28:
Hi @davidanger 
Yes, as mentioned in the release note:
https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.3.0
Serial menu has been reviewed, now the menu define only if Serial feature is enable and if default `Serial` instance is define.
So, by default, only the `Serialx` linked to `Serial` is instantiate.

If in your code `Serial1` and `Serial6` are not defined this is because they are not linked to default `Serial` instance.
You can define them like this on top of your sketch:
```
HardwareSerial Serial1(USART1);
HardwareSerial Serial6(USART6);
```

or add in `build_opt.h` file:
`-DENABLE_HWSERIAL1 -DENABLE_HWSERIAL6`

fpistm, 2018-06-29:
I've reproduced your issue by deleting the:
`arm-none-eabi-gcc/6-2017-q2-update/arm-none-eabi/include/stdint.h`
This file is part of arm gcc toolchains and not linked to the core.
arm gcc  package has not been changed between 1.3.0 and 1.2.0.
Could you check if the file exist?
Maybe the file is corrupted, issue during update process from 1.2.0 to 1.3.0?
I advice you to remove core and then reinstall it. Maybe armgcc package extraction had an issue...

davidanger, 2018-06-29:
Thanks @fpistm 
I uninstall the core lib and reinstall again this solved!~

fpistm, 2018-06-29:
Probably an issue during arm gcc archive  extraction. Archive was good else you would have crc issue raised.

davidanger, 2018-06-30:
Thank you very much @fpistm , this can be closed.

closed 2018-06-30 by fpistm