stm32duino/Arduino_Core_STM32 #274: Enhance GPIO usage using LL

fpistm, 2018-06-29:
Using LL instead of HAL will increase GPIO access.

davidanger, 2018-07-10:
Good News

ktand, 2018-11-02:
> Just an idea: In the Marlin STM32 HAL we used the register access to speed
> up the GPIO operations:
>
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_STM32/fastio_STM32.h
>
> https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_STM32/fastio_STM32.c
>
> For that we had to populate a pin mapping. Maybe the enhancement of the
> GPIO functionality could make this easier by providing such a mapping?

The link to the 2nd fille doesn't work. Should be:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_STM32/fastio_STM32.cpp

hasenbanck, 2018-11-02:
Upps, yeah. Corrected. Was just programming C stuff and brain got a blackout.

fpistm, 2018-11-05:
I've made a first try about this issue, see #368 
@hasenbanck , @ktand thanks for sharing.
I guess the `get_GPIO_Port` enhancement will allow you to avoid use `FastIO_init`

PR #368 (2018-11-16): GPIO enhancement
Fix #274 

* Clean up the digital code part
* Use LL instead of HAL for read/Write
* Add `digitalToggle `API
* Add `digital(Read|Write|Toggle)Fast` API (using PinName PY_n)
* Enhance `get_GPIO_Port` (macro instead of a function)
closed 2018-11-16 by fpistm