stm32duino/Arduino_Core_STM32 #275: Extend GPIO mode

fpistm, 2018-06-29:
Current mode available using pinMode() are:
 * `INPUT`
 * `INPUT_PULLUP`
 * `INPUT_PULLDOWN`
 * `OUTPUT`

Some other mode could be added as done in [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/9d46a1c27d4dc9dd4df06b76a1d81684519d8acc/STM32F4/cores/maple/io.h#L46)
At least:
 * `INPUT_ANALOG`
 * `INPUT_FLOATING`
 * `OUTPUT_OPEN_DRAIN`

Maybe AF mode.

Adminius, 2018-07-09:
Hi @fpistm ,
it sounds interesting, but this will make STM32 core incompatible to Arduino (if anybody will port STM32 sketch to Arudino)...

`INPUT_ANALOG` is the "same" as analogRead
`INPUT_FLOATING` is the same as INPUT
`OUTPUT_OPEN_DRAIN` ist the same as OUTPU LOW + INPUT...

IMHO: only INPUT_PULLDOWN makes sense.

fpistm, 2018-07-09:
I think no, it will not make core incompatible this is just an extension. Default Arduino value are kept.
If user use it yes his sketch will be incompatible but if he use it this is probably to take benefits of STM32 features. Up to him to tag it as STM32 compatible only.

davidanger, 2018-07-10:
What is different from INPUT, INPUT_ANALOG and INPUT_FLOATING?

davidanger, 2018-07-10:
I think add one more `OUTPUT_PULLUP` (MODER:01 OTYPER:1 PUPDR:01) OpenDrain + Pullup maybe have some useful in some cases. Can make I2C or something like that.

fpistm, 2018-07-10:
Right.
Why not add some new OUTPUT_* value.
Currently for some specific feature the config is get from the PeripheralPins.c
ex for ADC pin:
` {PC_2, ADC1, STM_PIN_DATA_EXT(STM_MODE_ANALOG, GPIO_NOPULL, 0, 3, 0)}, // ADC1_IN3`

where STM_PIN_DATA is used to get pin configuration coded on 32 bits:
STM PIN data as used in pin_function is coded on 32 bits as below:
```
     [2:0]  Function (like in MODER reg) : Input / Output / Alt / Analog
        [3]  Output Push-Pull / Open Drain (as in OTYPER reg)
     [5:4]  as in PUPDR reg: No Pull, Pull-up, Pull-Donc
     [7:6]  Reserved for speed config (as in OSPEEDR), but not used yet
   [14:8]  Alternate Num (as in AFRL/AFRG reg)
 [19:15]  Channel (Analog/Timer specific)
      [20]  Inverted (Analog/Timer specific)
      [21]  Analog ADC control - Only valid for specific families
 [32:22]  Reserved

```
See https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/PinNamesTypes.h#L36

PR #369 (2018-11-16): LL used for GPIO pin configuration
First implemention using LL instead of HAL for GPIO configuration.
Tests are on going.
Will fix #275 
Note: based on #368
commit 65554c11 (2018-11-16): Extend GPIO mode

Fix #275
closed 2018-11-16 by fpistm