stm32duino/Arduino_Core_STM32 #277: STM32F7 Interrupt

Vicky-S, 2018-07-09:
Is interrupt available for STM32F7. How can i initialize and use it?
Can i use any pin as interrupt or is there any dedicated pins for interrupt?

fpistm, 2018-07-09:
Hi @Vicky-S 
Yes interrupts are available.
All GPIO provide EXT interrupt.
one per pin number, port is not take in account.
So If you use PC0, EXTI0 will be used, then you cannot configure another interrupt on PC0.

Note that the function `attachInterrupt` also accept `std:function<void(void)>` in order to bind object method to an interrupt.

Here an example:

```
//#define USE_STD

volatile int state = HIGH;
void bt_handler(void) {
  state = !state;
}

#ifdef USE_STD
std::function<void(void)> bt_handler_f = bt_handler;
#endif

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
#ifdef USE_STD
  attachInterrupt(USER_BTN, bt_handler_f, FALLING);
#else
  attachInterrupt(USER_BTN, bt_handler, FALLING);
#endif  
}

void loop() {
  digitalWrite(LED_BUILTIN, state);
}
```

Vicky-S, 2018-07-09:
Thanks. Nice to have interrupt in STM32f7 on arduino. I wondered how to use it because different arduino architectures varies with interrupt intialization So i want to know about STM.

Should i define the pin number like arduino as 1 or A1, or do i need to name it as PA1 or PC1 like that?

Thanks for the example it would be helpful if u provide the detachInterrupt() syntax too.

Thanks for the quick response.

Adminius, 2018-07-09:
PA1/PC1 are better. (But 1 and A1 would also work)

fpistm, 2018-07-09:
Syntax is the same than Arduino API.
Only extensions is that PXy naming is allowed.

https://www.arduino.cc/reference/en/language/functions/external-interrupts/detachinterrupt/

Simply give the pin to detach interrupt.

fpistm, 2018-07-09:
Same as DUE which is based on STM32, that's why it use a pin as arguments.

Vicky-S, 2018-07-09:
Ok. Thanks.

fpistm, 2018-07-11:
Is it ok to close this?

Vicky-S, 2018-07-12:
Yes Please. It was working. If there is any problem i will let u know.

closed 2018-07-12 by fpistm