stm32duino/Arduino_Core_STM32 #279: RAK811 probkema with radio LoRa

sabas1080, 2018-07-15:
Hi 
I'm trying to get the Lora module working inside the board RAK811,

With the LMIC library, which apparently works with STM32 Murata CMWX1ZZABZ-078
https://github.com/mcci-catena/arduino-lmic

I had to comment the line
https://github.com/mcci-catena/arduino-lmic/blob/master/src/hal/hal.cpp#L247

But now I have an error in the line
https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/oslmic.c#L43

it seems not to start the radio

My sketch is

```
#include <SPI.h>
#include <lmic.h>
#include <hal/hal.h>

// LoRaWAN NwkSKey, network session key
// This is the default Semtech key, which is used by the prototype TTN
// network initially.
static const PROGMEM u1_t NWKSKEY[16] = {0x34,0x56,0x78,0x90,0x98,0x76,0x54,0x34,0x56,0x78,0x90,0x98,0x76,0x54,0x32,0x34};

// LoRaWAN AppSKey, application session key
// This is the default Semtech key, which is used by the prototype TTN
// network initially.
static const u1_t PROGMEM APPSKEY[16] = {0x56,0x78,0x90,0x98,0x76,0xFF,0xFF,0xF4,0x56,0x78,0x90,0x98,0x76,0x54,0x45,0x67};

// LoRaWAN end-device address (DevAddr)
// See http://thethingsnetwork.org/wiki/AddressSpace
static const u4_t DEVADDR = 0x23456789 ; // <-- Change this address for every node!

// These callbacks are only used in over-the-air activation, so they are
// left empty here (we cannot leave them out completely unless
// DISABLE_JOIN is set in config.h, otherwise the linker will complain).
void os_getArtEui (u1_t* buf) { }
void os_getDevEui (u1_t* buf) { }
void os_getDevKey (u1_t* buf) { }

static uint8_t mydata[] = "Hello, world!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 20;

// Pin mapping

const lmic_pinmap lmic_pins = {
    .nss = RADIO_NSS,
    .rxtx = RADIO_RF_CRX_RX,
    .rst = RADIO_RESET,
    .dio = {RADIO_DIO_0, RADIO_DIO_1, RADIO_DIO_2},
    .rxtx_rx_active = 1,
    .rssi_cal = LMIC_UNUSED_PIN,
    // the Murata module is direct-wired, we can use 8 MHz for SPI.
    .spi_freq = 8000000
};

void onEvent (ev_t ev) {
    Serial.print(os_getTime());
    Serial.print(": ");
    switch(ev) {
        case EV_SCAN_TIMEOUT:
            Serial.println(F("EV_SCAN_TIMEOUT"));
            break;
        case EV_BEACON_FOUND:
            Serial.println(F("EV_BEACON_FOUND"));
            break;
        case EV_BEACON_MISSED:
            Serial.println(F("EV_BEACON_MISSED"));
            break;
        case EV_BEACON_TRACKED:
            Serial.println(F("EV_BEACON_TRACKED"));
            break;
        case EV_JOINING:
            Serial.println(F("EV_JOINING"));
            break;
        case EV_JOINED:
            Serial.println(F("EV_JOINED"));
            break;
        case EV_RFU1:
            Serial.println(F("EV_RFU1"));
            break;
        case EV_JOIN_FAILED:
            Serial.println(F("EV_JOIN_FAILED"));
            break;
        case EV_REJOIN_FAILED:
            Serial.println(F("EV_REJOIN_FAILED"));
            break;
            break;
        case EV_TXCOMPLETE:
            Serial.println(F("EV_TXCOMPLETE (includes waiting for RX windows)"));
            if(LMIC.dataLen) {
                // data received in rx slot after tx
                Serial.print(F("Data Received: "));
                Serial.write(LMIC.frame+LMIC.dataBeg, LMIC.dataLen);
                Serial.println();
            }
            // Schedule next transmission
            os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);
            break;
        case EV_LOST_TSYNC:
            Serial.println(F("EV_LOST_TSYNC"));
            break;
        case EV_RESET:
            Serial.println(F("EV_RESET"));
            break;
        case EV_RXCOMPLETE:
            // data received in ping slot
            Serial.println(F("EV_RXCOMPLETE"));
            break;
        case EV_LINK_DEAD:
            Serial.println(F("EV_LINK_DEAD"));
            break;
        case EV_LINK_ALIVE:
            Serial.println(F("EV_LINK_ALIVE"));
            break;
         default:
            Serial.println(F("Unknown event"));
            break;
    }
}

void do_send(osjob_t* j){
    // Check if there is not a current TX/RX job running
    if (LMIC.opmode & OP_TXRXPEND) {
        Serial.println(F("OP_TXRXPEND, not sending"));
    } else {
        // Prepare upstream data transmission at the next possible time.
        LMIC_setTxData2(1, mydata, sizeof(mydata)-1, 0);
        Serial.println(F("Packet queued"));
    }
    // Next TX is scheduled after TX_COMPLETE event.
}

void setup() {

    
    pinMode(LED_BUILTIN,OUTPUT);//PB_6  //CRF3
    delay(200);

    Serial.begin(115200);
    while(!Serial);
    Serial.println(F("Starting"));

    
    digitalWrite(LED_BUILTIN,LOW);
    delay(500);
    digitalWrite(LED_BUILTIN,HIGH);
    delay(500);
    SPI.setMISO(RADIO_MISO);
    SPI.setMOSI(RADIO_MOSI);
    SPI.setSCLK(RADIO_SCLK);
    SPI.setSSEL(RADIO_NSS);
    //pinMode(RADIO_XTAL_EN,OUTPUT);
 
  //pinMode(RADIO_RF_CBT_HF,OUTPUT); //PB_7  //CRF2 HF
  //pinMode(RADIO_RF_CTX_PA,OUTPUT);//PA_4  //CRF1 PA
  
    //digitalWrite(RADIO_XTAL_EN,HIGH);
  
 
  //digitalWrite(RADIO_RF_CBT_HF,HIGH);
  //digitalWrite(RADIO_RF_CTX_PA,HIGH);

    // LMIC init
    os_init();
    Serial.println("Init");
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();

    // Set static session parameters. Instead of dynamically establishing a session
    // by joining the network, precomputed session parameters are be provided.
    #ifdef PROGMEM
    // On AVR, these values are stored in flash and only copied to RAM
    // once. Copy them to a temporary buffer here, LMIC_setSession will
    // copy them into a buffer of its own again.
    uint8_t appskey[sizeof(APPSKEY)];
    uint8_t nwkskey[sizeof(NWKSKEY)];
    memcpy_P(appskey, APPSKEY, sizeof(APPSKEY));
    memcpy_P(nwkskey, NWKSKEY, sizeof(NWKSKEY));
    LMIC_setSession (0x1, DEVADDR, nwkskey, appskey);
    #else
    // If not running an AVR with PROGMEM, just use the arrays directly 
    LMIC_setSession (0x1, DEVADDR, NWKSKEY, APPSKEY);
    #endif

for (int channel=0; channel<72; ++channel) {
      LMIC_disableChannel(channel);
    }
  //SF TTN
/*  
      LMIC_enableChannel(0);
      LMIC_enableChannel(1);
      LMIC_enableChannel(2);  //904.3Mhz
      LMIC_enableChannel(3);
      LMIC_enableChannel(4);
      LMIC_enableChannel(5);
      LMIC_enableChannel(6);
      LMIC_enableChannel(7);
      LMIC_enableChannel(8); 
   */
  //Home
   /* 
      LMIC_enableChannel(48);
      LMIC_enableChannel(49);
      LMIC_enableChannel(50);
      LMIC_enableChannel(51);
      LMIC_enableChannel(52);
      LMIC_enableChannel(53);
      LMIC_enableChannel(54);
      LMIC_enableChannel(55);
      LMIC_enableChannel(70);
      */
    // TTN defines an additional channel at 869.525Mhz using SF9 for class B
    // devices' ping slots. LMIC does not have an easy way to define set this
    // frequency and support for class B is spotty and untested, so this
    // frequency is not configured here.

    // Disable link check validation
    LMIC_setLinkCheckMode(0);

    LMIC_setAdrMode(false);
    
    // Set data rate and transmit power (note: txpow seems to be ignored by the library)
    //LMIC_setDrTxpow(DR_SF7,14);
    Serial.println("OK");
 
    // Start job
    do_send(&sendjob);
    
}

void loop() {
  os_runloop_once();
}
```

fpistm, 2018-07-15:
arduino-lmic is one of third party library I should investigate for LoRa to make it compatible with this core and vice versa.

sabas1080, 2018-07-16:
stop here someone did I'm going to try

http://www.stm32duino.com/viewtopic.php?f=48&t=3489

fpistm, 2018-07-16:
Right, forgot this while I've answered him :)

sabas1080, 2018-07-16:
I have also noticed that with version 1.3 of the core arduino it is not possible to communicate through SPI to module, I went back to version 1.2 and I can do it

fpistm, 2018-07-16:
This issue should be referenced in the #254.
This one i a request to support a third party library.
Anyway how do you setup the core and the new variant?

sabas1080, 2018-07-16:
SPI Core 1.2.0 work with library Lora https://github.com/sandeepmistry/arduino-LoRa, thus I discovered the error with this sketch

```
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);
  SPI.setMISO(RADIO_MISO_PORT);
  SPI.setMOSI(RADIO_MOSI_PORT);
  SPI.setSCLK(RADIO_SCLK_PORT);
  SPI.setSSEL(RADIO_NSS_PORT);
  LoRa.setPins(RADIO_NSS_PORT, RADIO_RESET_PORT, RADIO_DIO_0_PORT);

  /*SPIClass LoRaSPI(RADIO_MOSI_PORT, RADIO_MISO, RADIO_SCLK, RADIO_NSS);
  LoRa.setSPI(LoRaSPI);
  LoRa.setPins(RADIO_NSS, RADIO_RESET, RADIO_DIO_0);
*/
  Serial.println("LoRa Sender");

  if (!LoRa.begin(915E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}
```
with core 1.3.0 not working

fpistm, 2018-07-17:
I've just tested with Core 1.3.0 / DISCO L072CZ LRWAN / arduino-LoRa.
I have not issue, I can dump register and send packet.
I think there is a mistake in your setup and when you switch from one version to the other there something wrong.
So how do you use the core and the variant from #254 ?
Using git? Using standard install and then copying the variant?...

One note about the previous code:
```
  /*SPIClass LoRaSPI(RADIO_MOSI_PORT, RADIO_MISO, RADIO_SCLK, RADIO_NSS);
  LoRa.setSPI(LoRaSPI);
  LoRa.setPins(RADIO_NSS, RADIO_RESET, RADIO_DIO_0);
*/
```
`LoRaSPI `instance have to be global not local to the setup:

I've tested both possibilities:
Changing default SPI pins:
```
  SPI.setMISO(RADIO_MISO_PORT);
  SPI.setMOSI(RADIO_MOSI_PORT);
  SPI.setSCLK(RADIO_SCLK_PORT);
  SPI.setSSEL(RADIO_NSS_PORT);
  LoRa.setPins(RADIO_NSS_PORT, RADIO_RESET_PORT, RADIO_DIO_0_PORT); // set CS, reset, IRQ pin
```
and creating new instance:

```
SPIClass LoraSPI(RADIO_MOSI_PORT, RADIO_MISO_PORT, RADIO_SCLK_PORT, RADIO_NSS_PORT);

void setup() {
  ....
  LoRa.setSPI(LoraSPI);
  LoRa.setPins(RADIO_NSS_PORT, RADIO_RESET_PORT, RADIO_DIO_0_PORT); // set CS, reset, IRQ pin
}
```
both work fine.

fpistm, 2018-10-14:
Hi @sabas1080  any update on this ?

janbbeck, 2019-01-14:
I seem to have the same, or similar problem. I do have the DISCO L072CZ LRWAN, just as you do and it is working fine. However, on the RAK 811 Tracker, the code gets stuck in the 
LoRa.endPacket();

Any suggestions as to how to track down the issue?

> I've just tested with Core 1.3.0 / DISCO L072CZ LRWAN / arduino-LoRa.
> I have not issue, I can dump register and send packet.
> I think there is a mistake in your setup and when you switch from one [...]

fpistm, 2019-01-14:
Hi @janbbeck 
You can try to debug using Eclipse + Sloeber:

fpistm, 2019-02-12:
Close this issue as no update since several month from OP.

closed 2019-02-12 by fpistm