stm32duino/Arduino_Core_STM32 #284: NUCLEO_L4R5ZI

Lannakin, 2018-07-24:
Note that the variant files I've worked on aren't working yet, probably because I'm a total newbie and essentially have no clue what I'm doing.  I can almost completely guarantee that foolish errors will be present in any code I've touched.

Some blank space formatting will probably still be really inconsistent due to my inexperience and working on these from 2 different computers with different operating systems and just plain carelessness.

https://github.com/Lannakin/STM32-variants

fpistm, 2018-07-26:
Hi @Lannakin,
Thanks for your work, I've made a PR based on your work.
Feel free to comment/tests.
I've only update the clock config and fix the analog pins descriptions.
I've also added the Nucleo-L4R5ZI-P board as it is very closed. 

I've tested several basics features:
I/O, Serial, ADC, SPI, I2C, RTC, Low Power

fpistm, 2018-07-27:
It is fixed in the master based on your feedback.
See #280 
https://github.com/stm32duino/Arduino_Core_STM32/commit/3c66de722951baacb106901cb5503eb41940e720

Lannakin, 2018-07-27:
If it's helpful and useful, I can work on making parts of the other variants.  As I've demonstrated, I don't know too much about what I'm doing and I'll probably leave a trail of errors in my wake, but it might help me learn more.  I wouldn't be able to really test anything since I've only got the Nucleos L432KC, L4R5ZI, and F446RE, but if it's a bigger help than a burden for me to put together the contents for the variant files, I'd love to pitch a hand.

fpistm, 2018-08-01:
Up to you to contribute, thanks ;)

closed 2018-08-01 by fpistm