stm32duino/Arduino_Core_STM32 #294: How to query the CPU variant used in the board variant

hasenbanck, 2018-08-06:
Is there currently any way to let the program using the core to query the currently used CPU with a macro definition?

Case in point:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/backtrace/unwmemaccess.cpp

Marlin sets for debugging reasons ranges based on the CPU. Right now I'm using the pretty general definition "STM32F7xx" to set the needed values. But this is too broad, since I need to at least know the whole CPU variant (STM32F765 in my case).

hasenbanck, 2018-08-06:
Sadly the product_line is not enough, since the 765 line has different variants with different memory sizes. I will try to use your suggested approach to get the information out of the linker script.

closed 2018-08-06 by hasenbanck

fpistm, 2018-08-06:
I think this is the best approach as each variant provide the right linker file which include all the required information aligned with the board.
Added new entry in boards.txt will overload this and could lead to some misalignment. (ex with VECT_TAB_OFFSET)

hasenbanck, 2018-08-06:
Yeah, you are most likely right. I will try to patch Marlin this evening.