stm32duino/Arduino_Core_STM32 #295: Ask NUCLEO-F767ZI

davidanger, 2018-08-08:
I have NUCLEO-F767ZI, and I'm trying to Added this board to Arduino_Core_STM32.
So from I known it needs add and modify (for basic supported):
```
boards.txt
/variants/NUCLEO-F767ZI
/variants/NUCLEO-F767ZI/ldscript.ld
/variants/NUCLEO-F767ZI/PeripheralPins.c
/variants/NUCLEO-F767ZI/PinNamesVar.h
/variants/NUCLEO-F767ZI/stm32f7xx_hal_conf.h
/variants/NUCLEO-F767ZI/variant.h
/variants/NUCLEO-F767ZI/variant.cpp
```
Is anything advice and guide?  This my first time to try this.

fpistm, 2018-08-08:
Hi @davidanger 
right this is the basic lists.
Here, a full (I hope) description of all steps to perform:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

fpistm, 2018-09-14:
Support added thanks #320

closed 2018-09-14 by fpistm