stm32duino/Arduino_Core_STM32 #296: Buffered read / writes for the EEPROM emulation

hasenbanck, 2018-08-09:
Right now the EEPROM emulation inside the core (stm32_eeprom.c/h) provides two functions for reading and writing to an EEPROM emulated via flash (size of a page).

Both functions are touching the whole page in each function call, which is especially wasteful when writing to the flash.

To stay backward-compatible, I would propose to add four new functions to the core library:

```C
void eeprom_buffer_fill();
void eeprom_buffer_flush();
uint8_t eeprom_buffered_read_byte(const uint16_t __p);
void eeprom_buffered_write_byte(uint16_t __p, uint8_t __value);
```

The developer would then to first fill the static, private buffer with the fill function, interact with it and when he wants to save the content of the buffer to flash, flush the buffer to flash.

Implementing these changes shouldn't take much effort, since all pieces are already in place. I would like to do the implementation.

Comments? Suggestions?

fpistm, 2018-08-09:
The EEPROM is not really optimized. 
This feature is mainly to provide easiest way to save/restore some user settings which don't required high data rate and access else a real storage (SD,...) should be used.
Anyway, your suggestions seems a good improvement.

hasenbanck, 2018-08-09:
Since I'm currently adapting Marlin for your core, this problems popped up. Currently Marlin would safe the configuration (around 600 bytes) one byte at a time and would use 600 erase cycles of the whole page just for one save operation.

I implemented the usage of the Backup SRAM as an alternative, but I want to correct the current default operation mode non the less.

I will open a PR once I finished and tested the implementation.

commit e9dbcd31 (2018-09-21): [EEPROM emulation] Rework for buffered access

The current EEPROM emulation didn't had any buffered access to the
data on the flash. Every access to a byte would trigger a whole
page read / write. For writes it would mean, that the whole page
would be erased. For backward compability we will keep
the current functions as they are, but will add functions
to fill/flush the buffer and read from it.

The buffer size is static and is one page in size.

Fixes: #296
closed 2018-09-21 by fpistm