stm32duino/Arduino_Core_STM32 #298: I2C hardware or Software not working on STM32F103C8T6 with "generic STM32F103 series" selected

nikisalli, 2018-08-09:
using a normal I2C scanner code, i keep getting random addresses detected. i tried with another core (roger's core) with the same hardware and code and it works but when i switch to the stm32duino core it doesn't work anymore. i tried both hardware i2c and software i2c with the same results.

code:
`#include <Wire.h>

void setup() {
  Serial1.begin(115200);
  Wire.begin();
  Serial1.println("\nI2C Scanner");
}

void loop() {
  byte error, address;
  int nDevices;

  Serial1.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.

    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    
    if (error == 0) {
      Serial1.print("I2C device found at address 0x");
      if (address < 16) 
        Serial1.print("0");
      Serial1.println(address, HEX);

      nDevices++;
    }
    else if (error == 4) {
      Serial1.print("Unknown error at address 0x");
      if (address < 16) 
        Serial1.print("0");
      Serial1.println(address, HEX);
    }    
  }
  if (nDevices == 0)
    Serial1.println("No I2C devices found");
  else
    Serial1.println("done");

  delay(5000);           // wait 5 seconds for next scan
}`

fpistm, 2018-08-09:
Linked #217

nikisalli, 2018-08-09:
Isn't there a temporary fix i can use? Because from what i see it's like six months it is broken

LMESTM, 2018-08-10:
@nikisalli Hi - please find a proposed patch to fix the scan issue - let me know if you can test on your side.

nikisalli, 2018-08-10:
@LMESTM i just tested your patch on the bluepill (stm32f103c8t6) and it works flawlessly

LMESTM, 2018-08-10:
@nikisalli thanks for the quick feedback and testing !