stm32duino/Arduino_Core_STM32 #310: OT: STM32CubeProgrammer windows DFU driver issue

edogaldo, 2018-08-27:
Dear @fpistm please apologise for the OT issue..
I'm writing here to ask you, as long as you are an STM employee, if you could help addressing internally the subject issue.
Essentially the DFU driver delivered with the STM3CubeProgrammer (even with new version 1.1) is not signed so it's not possible to install it if the computer is locked to prevent installation of unsigned drivers..
There is also following ST support thread to address this issue: https://community.st.com/s/question/0D50X00009XkXagSAF/dfu-usage-with-stm32cbeprogrammer
but so far it received no attention at all from STM support...

Hope you can help, anyway thank you very much in any case.

Best regards, Edogaldo.

fpistm, 2018-09-03:
Hi @edogaldo,
No worries.
I will raised it internally.
Anyway, I will close this issue as it is not related to this project. The best way, is to use ST community as primary support.
Note that I would like unify tools used to flash the board (stlink, serial,...), as STM32CubeProgrammer is now multi os it is probably a good candidate, anyway I currently met some issue with the CLI...

fpistm, 2018-09-13:
Hi @edogaldo,
FYI, the STM32CubeProgrammer 1.2.0 is released and should contains signed DFU driver
https://www.st.com/en/development-tools/stm32cubeprog.html

closed 2018-09-13 by fpistm

edogaldo, 2018-09-13:
Thank you very much Frederic, I'm pretty sure your sponsorship has
been decisive!! ;)

fpistm, 2018-09-13:
In fact no, issue was already opened and resolution on going when I've asked ;)