stm32duino/Arduino_Core_STM32 #311: Problem with serial buffer size.

borombo96, 2018-08-31:
Hi, I've a problem with the serial of STM32L053R8 that is connected with a SIM-7000.
In the final application, I'll use also the expansion board I-Nucleo-LRWAN1.
For now I'm only trying to read the gps data from the sim, but I have problems with serial.
I also opened a post on forum.
When the chip is connecting to the satellite I read this in the serial:

AT+CGNSINF 

+CGNSINF: 0,,,,,,,,,,,,,,,,,,,,

OK

comand send
AT+CGNSINF 

+CGNSINF: 1,0,,,,,,,0,,,,,,1,,,,49,,

OK

comand send

ant it's ok.. but when it connects to satellite I read on the serial this:

comand send
AT+CGNSINF 

+CGNSINF: 1,1,20180821145743.000,38.121140,15.666
comand send
AT+CGNSINF 

+CGNSINF: 1,1,20180821145745.000,38.121241,15.665
comand send
AT+CGNSINF 

+CGNSINF: 1,1,20180821145747.000,38.121229,15.665

I don't understand why the data after +CGNSINF are not complete, in fact it should be something like:

AT+CGNSINF
+CGNSINF: 1,1,20180821151003.000,38.121095,15.666267,80.200,0.00,0.0,1,,0.7,1.0,0.7,,18,10,,,46,,

The code that I'm using is this:
`HardwareSerial SerialForSim7000(PA_10, PA_9);
 
HardwareSerial SerialLora(PC_11, PC_10); 
void setup()  
{
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  SerialForSim7000.begin(9600);
    SerialForSim7000.write("AT+CGNSCFG=1 \r\n");
  delay(500);
  SerialForSim7000.write("AT+CGNSPWR=1 \r\n");
  delay(500); 

}

void loop() // run over and over
{
SerialForSim7000.write("AT+CGNSINF \r\n");
while(SerialForSim7000.available())
Serial.write(SerialForSim7000.read());
Serial.println();
Serial.write("comand send");
Serial.println();
delay(2000);

  
}`
Ty 4 help everyone. :)

jmchiappa, 2018-09-03:
Hi borombo96,

I've also encountered this issue few months ago. Please try to increase buffer size in the hardwareSerial.h file:
`
#if !defined(SERIAL_TX_BUFFER_SIZE)
#define SERIAL_TX_BUFFER_SIZE 64
#endif
#if !defined(SERIAL_RX_BUFFER_SIZE)
#define SERIAL_RX_BUFFER_SIZE 64
#endif
`
Replace 64 bytes by something like 1024... and it should work.

borombo96, 2018-09-04:
I have his in hardwareSerial.h
`#if !defined(SERIAL_TX_BUFFER_SIZE)
#if ((RAMEND - RAMSTART) < 1023)
#define SERIAL_TX_BUFFER_SIZE 16
#else
#define SERIAL_TX_BUFFER_SIZE 64
#endif
#endif
#if !defined(SERIAL_RX_BUFFER_SIZE)
#if ((RAMEND - RAMSTART) < 1023)
#define SERIAL_RX_BUFFER_SIZE 16
#else
#define SERIAL_RX_BUFFER_SIZE 64
#endif
#endif`

I try to change 64 with 1024 and also 16 with 1024 but it doesn't work. Ty 4 help

fpistm, 2018-09-04:
This is not the right code. I guess you try to edit AVR core code not STM32.
This is the AVR HardwareSerial.h:
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/HardwareSerial.h#L42
STM32 one:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/HardwareSerial.h#L42

fpistm, 2018-09-08:
@borombo96 
is this ok to close as you mention it as solved on the forum?

borombo96, 2018-09-08:
Yes, thank u a lot for help. ☺️

fpistm, 2018-09-08:
Welcome.

closed 2018-09-08 by fpistm