stm32duino/Arduino_Core_STM32 #314: STM32L0xx_HAL_CONF.h defines not correctly supported in core.

disk91, 2018-09-04:
When you want to make your own board definition and modify the STM32L0xx_HAL_CONF.h to remove unsupported or unexisting feature it creates dependency failure. As an exemple if you try to compile w/o SPI
//#define HAL_SPI_MODULE_ENABLED
you get:

in file included from /Users/paul/Documents/Arduino/hardware/stm32/1.3.0/cores/arduino/board.h:19:0,
                 from /Users/paul/Documents/Arduino/hardware/stm32/1.3.0/cores/arduino/wiring.h:41,
                 from /Users/paul/Documents/Arduino/hardware/stm32/1.3.0/cores/arduino/Arduino.h:32,
                 from /var/folders/pz/tbk987c56x3791b90cqz65n80000gn/T/arduino_build_470787/sketch/Blink.ino.cpp:1:
/Users/paul/Documents/Arduino/hardware/stm32/1.3.0/cores/arduino/stm32/spi_com.h:53:5: error: 'SPI_HandleTypeDef' does not name a type
     SPI_HandleTypeDef handle;

Basically spi_com.h should not be refered if the SPI module is unactivated.
We have the same issue with I2C.

ADC, RCC, we can't compile w/o these Modules too.

fpistm, 2018-09-05:
Hi @disk91,
Disable HAL feature will be mainly covered by #228.
All features linked to a built-in library, if not included, code is not embedded but it still the related PinMap_<features> arrays 
The only one which will be hard to fully removed will be RCC as it is requested to configure and enable clocks. LL have to be used to remove it.

fpistm, 2018-10-12:
Added RCC to the list in #228.
Closed this one as it is a duplicated issue

closed 2018-10-12 by fpistm