stm32duino/Arduino_Core_STM32 #315: wiring.h is refering Tone.h but not using it

disk91, 2018-09-04:
We have in wiring.h an unneeded dependency on Tone.h. I think we can remove it.

fpistm, 2018-09-05:
Arduino Tone examples does not include `Tone.h`
It has to be included by the core.
For samd it is included in `Arduino.h` before the `wiring.h` but this do the same.

disk91, 2018-09-12:
Lol ... Tone examples should be fixed ... arduino code style ! damned

fpistm, 2018-09-13:
You can submit an issue to Arduino :)
As I said Tone api are not declared the same way across arduino core. One time in a dedicated include file on other in Arduino.h.
Anyway, if tone api is not called in the code this is removed form the binary so it no memory space consumed for it.
If you are ok, could you close this issue ?

fpistm, 2018-09-26:
As no feedback and that include of `Tone.h` will not be removed while official Arduino example does not include it, I close this issue.

closed 2018-09-26 by fpistm