stm32duino/Arduino_Core_STM32 #316: Which timer to use for incremental (aka quadrature) encoder

jmchiappa, 2018-09-05:
Hi @fpistm ,

I plan to write a library using the encoder timer feature (HAL_TIM_Encoder_Init, HAL_TIM_Encoder_Start_IT). This library should work with freertos, servos, PWM, etc.
Please could you advise which TIMER I must use (i.e. which is not in conflict with other features).
For now, I have NUCLEO L476RG only but it should work with all NUCLEO at the end.
Maybe this lib already exists :) ?
BTW, it should be great to have a big picture of timer resources (which timer is using for SERVO, STEPPER, FREERTOS, PWM, and so on).

Thank you

fpistm, 2018-09-05:
Hi @jmchiappa,
currently the only ones used are the one defined in the variant.h of the variant.
Example for Nucleo L476RG:
```
// Timer Definitions
// Do not use timer used by PWM pins when possible. See PinMap_PWM.
#define TIMER_TONE              TIM6

// Do not use basic timer: OC is required
#define TIMER_SERVO             TIM2  //TODO: advanced-control timers don't work
```

In the coming month, I've planned to write an HardwareTimer library offering API to use timers features.

jmchiappa, 2018-09-05:
Thank you for your quick reply.

Refering to the pinMAP _PWM, it sounds that more than theses ones can be used.
All pins defined as PWM could use a timer. Timers that can be used are:
TIM1, TIM2, TIM3, TIM4 and TIM8 (for L476RG) + TIM6 (TIMER_TONE)
Am I right ?
Should I not use all of these or could you ensure that using one of them will no introduce some side effects ?

fpistm, 2018-09-05:
Yes, anyway some other timers could be used:
TIM5, TIM15, TIM16, TIM17
pinMAP _PWM is just a basic/random setup.
Ex:
Currently:
```c
//  {PB_14,  TIM15,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF14_TIM15, 1, 0)},  // TIM15_CH1
    {PB_14,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 2, 1)},  // TIM1_CH2N
//  {PB_14,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 2, 1)},  // TIM8_CH2N
```
but user can change the Timer linked to the pin to use TIM15 by changing comments:
```c
    {PB_14,  TIM15,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF14_TIM15, 1, 0)},  // TIM15_CH1
//  {PB_14,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 2, 1)},  // TIM1_CH2N
//  {PB_14,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 2, 1)},  // TIM8_CH2N
```
I've planned to add guard to prevent muti timer usage but I do not study deeply how to perform that.
Anyway, it's always up to the user to use a free peripheral.

fpistm, 2019-04-03:
Hi @jmchiappa 
any update on this ?

jmchiappa, 2019-04-04:
Hi Frédéric
I just hold this lib for a while. I hope to restart this dev soon. Unless new information are availables, I close this topic.
Thank you for your hard work

closed 2019-04-04 by jmchiappa

fpistm, 2019-04-04:
Welcome and thanks Jean-Marc.