stm32duino/Arduino_Core_STM32 #330: Request support for standard (ST) DFU bootloader upload

edogaldo, 2018-09-19:
As per title, it would be useful to allow uploading via the standard ST DFU bootloader (where available).
This could be a good compromise for boards like the Black F407VE which else would require an external tool for initial load..

ppescher, 2018-09-21:
I have done this in our OpenTracker fork.

See board/platform configuration files:
https://github.com/RoboTech-srl/opentracker-stm32-core/blob/7fbe63f592d32d0f7aac16dd4e6d4949937ddf4e/boards.txt#L46 (and below)
https://github.com/RoboTech-srl/opentracker-stm32-core/blob/7fbe63f592d32d0f7aac16dd4e6d4949937ddf4e/platform.txt#L150 (and below)

I am using STM32CubeProgrammer in DFU mode. I'm using this on Windows, there is still an issue with the wrong driver being loaded because the new one does not have a digital signature, however it can be manually switched in the Device Manager. I also briefly tested it on linux. Not sure about MacOSX, but I think I got the application path right.

The implementation of the boot call in this commit:
https://github.com/RoboTech-srl/opentracker-stm32-core/commit/61d98b5d0827fb6bc1650513b2d10ecf384fcfd3
Later modified in:
https://github.com/RoboTech-srl/opentracker-stm32-core/commit/83ca6ee908aa8845ebbd8b9538f860450794c7b9

In the first attempt I reserved a variable (boot flag) in the newly created ".noinit" section, otherwise it would have been zero-initialized by the reset handler.
In the second one I moved the checkBoot() call inside the reset handler, before zero-init. I left the ".noinit" section in the modified linker script, even if no longer required.
I thought it was less invasive change to not requiring all the linker scripts to be modified. Even if kind of hacking or hijacking the reset handler (I count on the fact that all reset handlers are equal and have defined the label LoopCopyDataInit).

And the "1200 baud trick" is in my USB CDC attempt:
https://github.com/RoboTech-srl/opentracker-stm32-core/blob/b6f0b430c4049bd801d27be074034f58cb5de6b7/cores/arduino/stm32/usbd_cdc_if.cpp#L171

If you think this is a proper solution I might prepare a pull request.

edogaldo, 2018-09-21:
> RoboTech-srl@61d98b5
> RoboTech-srl@83ca6ee

Thanks Paolo, just to let you know that v1.2.0 of the CubeProgrammer
provides a signed DFU driver (ref #310).

Cheers, E.

ppescher, 2018-09-22:
Cool, thanks Edoardo!
It would have been nice to let us know in the same post where we both wrote on the ST Community forum.
This is good news anyway.

edogaldo, 2018-09-22:
... :)

fpistm, 2018-10-10:
This will be handled thanks #351

chrissbarr, 2018-10-10:
I think the 1200 baud rate trick is a great idea @ppescher. 

Yeah, the dfu-util implementation will only handle the upload part, and needs the board to be in the right state - so a way to reset the board first would be needed. Boards with custom bootloaders should then be able to accept an upload after being reset (assuming they enter DFU mode for a short time, which is I think common behaviour a la Maple). Boards with only the built-in DFU bootloader will still need the BOOT0 pin set accordingly - I don't know that there's an easy way to use that bootloader without the user interacting with the hardware (though it [looks like it is possible](https://stm32f4-discovery.net/2017/04/tutorial-jump-system-memory-software-stm32/), but probably hard to generalise to all chips).

I think ideally the reset and upload methods should be separate - it's likely there are upload methods besides DFU that could be used on reset.

ppescher, 2018-10-10:
Hi @chrissbarr 
Yes it is definitely possible to call into the ROM bootloader to activate DFU. That's what I did (see my https://github.com/stm32duino/Arduino_Core_STM32/issues/330#issuecomment-423603895).

I used a similar approach to what micropython does (see https://github.com/micropython/micropython/blob/338635ccc64204b6f388cfaafca00e120090c622/ports/stm32/modmachine.c#L260) even though I jump to the bootloader after resetting the system (to make sure hardware is in the same state as if you used the BOOT0 pin).

fpistm, 2019-05-03:
Closed thanks #514

closed 2019-05-03 by fpistm