stm32duino/Arduino_Core_STM32 #341: Make usbd_interface_init() weak so that it can be redefined in sketch or external library

edogaldo, 2018-09-27:
Hello @fpistm, as I can understand function `usbd_interface_init()` is a hook to initiate USB function (HID at the moment) if required:
https://github.com/stm32duino/Arduino_Core_STM32/blob/4919790f67495ebdc3547aa1b96ade46967a178a/cores/arduino/main.cpp#L57-L59
correct?
If yes, what about making its definition 
https://github.com/stm32duino/Arduino_Core_STM32/blob/4e1c22561ae15c976a42699c97e08c53dbb3f4b7/cores/arduino/stm32/usb_interface.c#L119-L131
as `weak `so that users can implement USB features in sketches or external libraries?

Thanks in advance and best, E.

fpistm, 2018-09-28:
Yes (at this moment).
As mentioned in other Issuer/PR this will be removed as USB will be moved in builtin library.

edogaldo, 2018-09-28:
Thank you @fpistm.
Which are the planning for including UBS?
In my opinion it would anyway be better to embed it as an external library that can be included if needed in order to avoid useled code bloating if not needed.
Going back to my question and your answer, "Yes (at this moment)", does it mean you will change 
```
void usbd_interface_init(void)
{
  ...
}
```
to
```
__attribute__((weak))
void usbd_interface_init(void)
{
  ...
}
```
in the meantime?

Thanks, E.

fpistm, 2018-09-28:
I've answered here to @hasenbanck 
https://github.com/stm32duino/Arduino_Core_STM32/issues/222#issuecomment-425340621
What I would mean by "at this moment" is this part will be reviewed.

fpistm, 2018-10-12:
I guess this issue will be deprecated with #344

edogaldo, 2018-10-12:
Well, the aim of this issue is to request the possibility for the user to be able to define his own USB initialization procedure, possibly also at sketch level (and not only at variant level).
Any solution which goes in that direction is fine for me.

Cheers, E.

closed 2019-01-11 by fpistm