stm32duino/Arduino_Core_STM32 #342: How to use "build_opt.h"?

edogaldo, 2018-09-27:
As per subject. 

Thanks in advance, E.

fpistm, 2018-09-28:
Hi @edogaldo 
https://github.com/stm32duino/wiki/wiki/Customize-build
Is it ok with this?

edogaldo, 2018-09-28:
Thank you @fpistm, the wiki topic is clear and helpful; just a couple of comments:
- I'd suggest to add the name of the file in the title for better/faster identification, i.e. "How to customize build options - build_opt.h"
- Use of the ".h" extension is misleading: I was thinking of it as a normal header file and I was trying to add options in a pre-processor format ("#define xxx yyyy"); it would have been better to me to use a different extention (i.e. "build_opt.conf") anyway, once clarified, it's ok

Thanks again and best, E.

fpistm, 2018-09-28:
I used the .h extension because Arduino IDE do not allow to add all extension file you want. If i well remember only .h, .c, .cpp, .ino.
I've updated the title.

edogaldo, 2018-09-28:
Ah ok, didn't know.

Thanks you!

edogaldo, 2018-09-28:
Just update also above link because it's no more valid.

closed 2018-09-28 by edogaldo