stm32duino/Arduino_Core_STM32 #347: 3D printer board support

hasenbanck, 2018-10-01:
Since Marlin 2.0 will support many different MCU architectures beside the AVR, STM32 MCU support is also in the work. There is currently planning ongoing to have one HAL for all STM32 boards.

The first such board was REMRAM V1, but there are already other board on the market, which will need support for Marlin for this Arduino core (since this core is the only core, that supports all STM32 targets).

This issue should simply act as a preparation, for future board definition request that will come (and of which I will most likely write quite some).

I think it would be best if remove the "REMRAM" top-level category and add a general "3rd party boards" or "3D printer board" category, so that we can group these board better together.

For reference:
https://github.com/MarlinFirmware/Marlin/pull/11750 and https://github.com/MarlinFirmware/Marlin/issues/7076

hasenbanck, 2018-10-04:
Currently there are the following boards that would need an additional variant:
 * Chitu3D V3.9
 * Malyan M200 (There are various motherboard versions I think)
 * GTM32_PRO_VB
 * GTM32_MINI
 * STEVAL-3DP001V1

There is also the "PrntrBoard" by @ghent360, but that is using the STM32F446 NUCLEO-64 as a platform and is already supported.

@fpistm Do you might have access to the STEVAL-3DP001V1 board?

@Phr3d13 @xC0000005 Do you have some of the boards available? I could write the board variants if you could assist in testing them.

ktand, 2018-10-04:
I would like to add my board, "Armed", to that list. It's based on
STM32F407 and running Marlin 2.0.x with the STM32 core.

Karl

ghent360, 2018-10-04:
Are we doing the right thing?

If I compile Marlin for AVR, I don't have to choose every single RAMPS offspring board ever created in the Arduino IDE. Somehow they manage not to require a board variant for the plethora of AVR based printer controllers.

What are we trying to achieve that is different in this case, so creating a board configuration would actually be beneficial?

Wouldn't it be simpler if we create "generic" variants that correspond to popular CPU packages and number the pins after the actual pin numbers on the chip.

ktand, 2018-10-04:
I guess every board maps their PWM, I2C and SPI pins etc. differently,
probably requiring variants for every board.

Karl

hasenbanck, 2018-10-04:
Yes, we are indeed doing the right thing. AVR based boards most often pin compatible with their big Arduino cousin and that's why you often get compability out of the box.  As soon as you deviant from that (like the Rambo series of boards) you need also another board definition.

STM32 chips and boards are not that uniform. For example: I use a similar chip from the STM32F765 family then another custom board. But I can't use that variant definition, since even when we use the same CPU number,  we use different versions of it (I use the VI with 2 MB Flash). The other board also uses different PWM outputs and we have to use for that reason different timers (which could also lead to the need to user different SPI / I2C peripherals).

And since we have different boards out in the wild, which use different peripherals etc, we have to support them somehow (Since we can't change their silicon).

chrissbarr, 2018-10-05:
I'm working on two boards built around the STM32F446VET6: RUMBA32 & PicoPrint.

I have everything running decently with STM32GENERIC + Marlin 2.0.x + STM32F4 HAL, but I'm looking to swap them over to this core so I can move them towards the more unified STM32 Marlin HAL you've been working on @hasenbanck.

I've only started testing out this core yesterday, so not sure if I'll run into trouble yet - but in any case, I think a folder for 3D printer boards is a good idea, as it's sure to be a list that will grow fairly quickly.

xC0000005, 2018-10-05:
Seconding this with a real example. The M200 boards (STM32F107 and STM32F070) require different loading offsets than a bluepill because they have a proprietary bootloader, and require some different startup code (The STM32F103 startup code doesn’t load SP by default, because the processor would, but the bootloader doesn’t do it, so the startup code needs to). V1 has a pullup transistor on PB9 that’s required for USB. Sure we could use the generic variants, but that would involve changing their behaviors. Or we could add specific variants and let the bluepills continue to use PB9 for whatever they do while the M200s use it to enable USB, and so on. Variants are good, though they do come with some baggage.

xC0000005, 2018-10-05:
I believe I own nearly every revision of the M200 V1 and V2 boardsets, and I’d not only be happy to test, I can assist in the varianting when I’m at home instead of traveling.

hasenbanck, 2018-10-05:
@xC0000005 Great to hear. I will wait until you returned and looked into the matter yourself. Since you already wrote the variants for the STM32generic core, it should be faster when you do it yourself (my vacation is soon over, so my time budget will decrease fast soon).

Phr3d13, 2018-10-05:
I have both the gtm32 variants. I'd be happy to help whenever you need me.

hasenbanck, 2018-10-05:
@Phr3d13 Did you already use those boards with the old HAL with Marlin 2.0? I kinda need some documentation about the pin / peripheral configuration to start with. Do you happen to know if there is some documentation about these topics?

Phr3d13, 2018-10-05:
[Smartto](https://github.com/Geeetech3D/Smartto-IAR) - This is their firmware for both of their stm32 boards.
[GTM32 Pro VB](https://github.com/Geeetech3D/Diagram/blob/master/Rostock301/Hardware_GTM32_PRO_VB.pdf) - Circuit diagram for the PRO VB board
[GTM32 Mini](https://github.com/Geeetech3D/Diagram/blob/master/3DPrinter_A30_E180/Hardware_GTM32_MINI_S.pdf) - Circuit diagram for the Mini board
They don't have anything like a pins.h file anywhere. pins are defined in each relevant file, IE endstop pins are defined in Smartto-IAR/src/User/endstop.h, etc

hasenbanck, 2018-10-05:
I think those files and the information in the pin definition in Marlin
should be enough. I will look tomorrow into it!

Phr3d13, 2018-10-05:
Sweet! No rush. Thanks for allowing my request. ;)

fpistm, 2018-10-08:
> I think it would be best if remove the "REMRAM" top-level category and add a general "3rd party boards" or "3D printer board" category, so that we can group these board better together.
I guess it should be fine ;)

> @fpistm Do you might have access to the STEVAL-3DP001V1 board?
I have to check

ethanspencer, 2018-10-10:
3D printing is becoming popular these days. One of the most fundamental things required to make a DIY 3d printer is its printing skills and you can check the further tips here https://goo.gl/4JXq3G I also followed and now I have my printer with me.

ffnull, 2018-10-19:
@hasenbanck Im have STEVAL-3DP001V1 so i can help and be tester. What i need to do?

hasenbanck, 2018-10-19:
@ffnull I currently have little time, since I'm pretty busy with my side project RemRam and Heteromycin. So currently your best options would be: To write the variant yourself.

The board is pretty well documented, so you could look into STs fork of Marlin 1.10 and could spot their pins definitions / mappings [file](https://github.com/St3dPrinter/Marlin4ST/blob/master/stm32_cube/STM32/Drivers/BSP/STM32F4xx-3dPrinter/stm32f4xx_3dprinter_misc.c)

The way to create a variant is also pretty well documented in the [wiki](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)).

You kinda need to have a basic understanding how the STM32 MCU in general has to be configured using the STM32Cube MX tool though. It would require you to learn lot of new stuff potentially, but could be your fasted way to get your board supported.

The Marlin part later should be rather easy, since you have the old Marlin [pin definition](https://github.com/St3dPrinter/Marlin4ST/blob/master/stm32_cube/Marlin/pins_STM_3DPRINT.h) already and "only" need to port it to Marlin 2.0 using the variant definition you created. We currently have a STM32 HAL for F0, F1, F4 and F7 MCUs, so there shouldn't be any hard problems on your way. Only real missing feature as of right now is the CDC feature (Serial to USB), but that is beeing worked on as of right now.

ffnull, 2018-10-19:
@hasenbanck when i download latest brunch of bugfix 2.0 and opened it in platformio i cant compile env:STM32F4 just for example. to many errors with eeprom and timers. what i need to do to compile it and make my board work based on worked example?

ffnull, 2018-10-19:
@xC0000005 can you give short step by step?

xC0000005, 2018-10-19:
Follow the getting started steps here:
https://github.com/stm32duino/Arduino_Core_STM32 <https://github.com/stm32duino/Arduino_Core_STM32>

I’d set your board to a generic F4 until you can build a variant for it. You’ll need to be comfortable debugging it if something goes wrong.

ffnull, 2018-10-19:
@xC0000005 in platformio im set generic F4 but it whant compile. Will try your method in arduino IDE

ffnull, 2018-10-19:
@xC0000005 still cannot compile generic f4 in arduino too. many errors on serial and eeprom

ffnull, 2018-12-25:
@fpistm Im write my own board variant. xC0000005 tryes help. But its doesnt work correct. So im start from new page) Im write all pins. But mapping not correct. Board starts, UART works. So need help in small fixes

fpistm, 2019-02-26:
> 
> 
> > @fpistm Do you might have access to the STEVAL-3DP001V1 board?
> 
> FYI, I try to get one but don't know when...

@hasenbanck 
Finally got it! :grin:
I will do the variant when I got a slot

fpistm, 2019-02-26:
> 
> 
> @fpistm Im write my own board variant. xC0000005 tryes help. But its doesnt work correct. So im start from new page) Im write all pins. But mapping not correct. Board starts, UART works. So need help in small fixes

@ffnull sorry missed your message. Is it ok now? You can do a PR to share it and be able to review.

hasenbanck, 2019-02-26:
> @hasenbanck
> Finally got it! 😁
> I will do the variant when I got a slot

That took a long time! But great to hear!

I personally look right now at the STM32MP1 chip you guys just announced. Maybe I will design someting using it. But it's too early right now, for now I'm waiting for the dev boards.

BennehBoy, 2019-02-26:
Those chips look really nice.

fpistm, 2019-02-27:
> shoemakers children are the worst shod
😀

ffnull, 2019-02-27:
@fpistm Good. I will try to publish my attempts today

ffnull, 2019-02-27:
@fpistm Unfortunately. For reasons that are not clear to me, I managed to delete the entire folder with my projects. I lost my version of the board(((((((((((

fpistm, 2019-02-27:
Well, this confirm that using a versioning tools could avoid this kind of issue.

xC0000005, 2019-02-27:
Ffnull, there’s probably still a copy of your board in the hackaday.io Channel. Heck, I might have it somewhere.

ffnull, 2019-02-27:
@xC0000005 Thank You!
@fpistm here is my good comented attempts
[6986651545765103573.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/2912487/6986651545765103573.zip)

fpistm, 2019-02-28:
Thanks @ffnull  and @xC0000005 
So your variant is for this board: https://www.st.com/en/evaluation-tools/steval-3dp001v1.html ?
Do you have the updated board.txt?

ffnull, 2019-02-28:
Sorry. But no

ShenRen, 2019-02-28:
Hellow ! 
I am work with a 3D printer board project . by using TrueSTUDIO IDE ,  Arduino_Core_Stm32 + Marlin V2 Firmware. But I have encountered many serious bugs.

1 : UART RX Buffer Corruption in `HardwareSerial`
same as https://github.com/stm32duino/Arduino_Core_STM32/issues/421

2 : PWM will make gpio float or out of control
I think  the reason is that when every time we apply a new pwm value , the Core will  reinit something which made  gpio  out of control.

3 : ADC  value  float 
reason unknow

ask for help !!!

fpistm, 2019-02-28:
@ffnull, I've made the PR based on your work. I've made some clean and fixes.

NilsHasenbanck4758, 2019-02-28:
@ShenRen 
Please open separate issues for the bugs you found. I don't think the information you provided are enough, since we have installation of your setup (STM32 MCU, Marlin 2.0) that work very well and have no issues with the ADC / PWM. So you may need to provide code to help us reproduce the issues.

hasenbanck, 2019-02-28:
@ShenRen
Please open separate issues for the bugs you found. I don't think the information you provided are enough, since we have installation of your setup (STM32 MCU, Marlin 2.0) that work very well and have no issues with the ADC / PWM. So you may need to provide code to help us reproduce the issues.

ShenRen, 2019-03-01:
Ok ！ I will open a new issues  and upload my code this weekend.

ShenRen, 2019-03-04:
I've solved all the problems, by downloading the release program .
All the time , I just download the debug program to mcu for track the code . which makes the firmware  not function properly.

fpistm, 2019-05-29:
@ffnull,  @hasenbanck,
I've tested the basics features of the ST3DP0001 EVAL and it is ok.
I will merge #462

Gamester17, 2019-06-26:
Can I suggest that you help port Marlin 2.0 firmware to run the new STM32MP1 MPU series?
https://blog.st.com/stm32mp1-mpu-stm32mp157a-ev1-stm32mp157c-dk2/

I am only an end-user / technology enthusiast myself and I do not actually know if anyone is working on a 3D-printer controller board based on the new STM32MP1 MPU series or not, however, I read that STMicroelectronics claims STM32MP1 architecture enables developers to use the same software as STM32 MCU series so porting Marlin 2.0 firmware to STM32MP1 MPU series should be possible in theory as it already supports STM32, and theoretically, with the STM32MP1 MPU series you could run both Marlin and OctoPrint or similar Linux distribution on the same board.

For technical specification summery checkout:
https://www.cnx-software.com/2019/02/21/stmicro-stm32mp1-cortex-a7-m4-mpu/

STM32MP157A-DK1 and STM32MP157C-DK2 Discovery kits are $99 with LCD or $69 without
https://www.st.com/en/evaluation-tools/stm32mp157a-dk1.html
https://www.st.com/en/evaluation-tools/stm32mp157c-dk2.html
STM32MP157A-EV1 Evaluation kit for PCB engineers does have a steeper price at $399
https://www.st.com/en/evaluation-tools/stm32mp157a-ev1.html

First read about this MCU/MPU in the release news about PanGu board from I2Som
https://www.cnx-software.com/2019/06/25/pangu-board-stm32mp1-sbc/

fpistm, 2019-10-28:
As there is more and more support request for new variant, all are grouped in a single issue to ease tracking: #722.

closed 2019-10-28 by fpistm