stm32duino/Arduino_Core_STM32 #360: upload vect

Homayoun-MH, 2018-10-26:
Please, before reporting any issue:
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported or discussed on the [stm32duino forum](http://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the Arduino_Core_STM32.
- **Avoid** to submit a GitHub issue for project troubleshooting.
 
Any questions/feedback/sugestions should be discussed on the [stm32duino forum](http://stm32duino.com):
  * questions on the [STM32 Core](http://stm32duino.com/viewforum.php?f=48)
  * bugs/enhancements on the [STM core: Bugs and enhancements](http://stm32duino.com/viewforum.php?f=49)

When reporting any issue, please try to provide all relevant information:

- Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)
- Which STM32 based board you used
- If other hardware is required to reproduce the issue, please provide links or part numbers where this hardware can be purchased.
- Which version of STM32 core and Arduino, Tools menu settings, operating system, ...

Thanks

fpistm, 2018-10-26:
Hi this is currently not supported, this will comes with DFU upload method.
The VECT_TAB_OFFSET can be redefined but the linker script need to be updated

Homayoun-MH, 2018-10-26:
Thanks

but is there any way to get USB serial port working?

fpistm, 2018-10-26:
This is plan for the 1.5.0 release:
https://github.com/stm32duino/Arduino_Core_STM32/pull/344

fpistm, 2019-05-03:
`VECT_TAB_OFFSET` can be redefined thanks `build_opt.h` if required.
It will be also handle in #411 and will also covered the upload part.

closed 2019-05-03 by fpistm