stm32duino/Arduino_Core_STM32 #363: High Flash Usages for any program

mayurharge, 2018-10-31:
I am using the latest arduino core stm32. On a windows machine. 
I am trying to compile a code for nucleo32 board with  L031K6 mcu. Strangely The compiled code size is becoming very high and so i am unable to include any library. 
I tried to compile basic analogtoserial code from arduino examples. That code alone is using almost 64% of the total flash, if i add any library it son reaches to 90% or sometimes beyon 100%. 

I checken on stmduino forums and some people suggested about adding `--specs=nano.specs` in the end of line beginning with `"recipe.c.combine.pattern=...."` in platforms.txt but in my case this is already there. 

I have tried many things but it seems like an issue with the core. Can someone point me in right direction here...

fpistm, 2018-10-31:
Hi @mayurharge 
There is no issue, some actions are on going to decrease the size.
The next release (will come today 1.4.0) will decrease the basic size.
You could also fully disabling the Serial feature if you do not use it thanks the menu.
Other actions will be to use the LL for the clock config and also be able to disable useless HAL See #228

mayurharge, 2018-10-31:
thanks, @fpistm
I will try out disabling useless HAL and will wait for next release

fpistm, 2018-11-02:
New release done.
For useless HAL this  will required  #228.
Note also that LL will be more deployed to reduce size.

closed 2018-11-13 by fpistm