stm32duino/Arduino_Core_STM32 #370: Need help in SPI.

Vicky-S, 2018-11-10:
I got successfully programmed STM32F103 and I can able to do basic tasks.

Now I need help in using 2 SPI in STM32F103 at the same time. The STM32F103 has 2 SPI. Which is the default one uses Arduino SPI library and how can I initialize another SPI?

I need to access 2 SPI at the same program. I have tried to parallel the SPI devices in single library but its not working well. That's why I try to use 2 SPI.

fpistm, 2018-11-10:
By default `SPI `instance are mapped on SPI1 using:
```
#define PIN_SPI_SS              PA4
#define PIN_SPI_MOSI            PA7
#define PIN_SPI_MISO            PA6
#define PIN_SPI_SCK PA5
```
See https://github.com/stm32duino/Arduino_Core_STM32/blob/0dd2047fbf7d7d93f192d75c81068aced9450161/variants/BLUEPILL_F103C8/variant.h#L99

SPI2 is available on pin PB15/PB14/PB13. 
See here https://github.com/stm32duino/Arduino_Core_STM32/blob/0dd2047fbf7d7d93f192d75c81068aced9450161/variants/BLUEPILL_F103C8/PeripheralPins.c#L186

To use the second one:
`SPIClass SPITwo(PB15, PB14, PB13);`
Then use it as you used the default `SPI `istance.
Note that you can't use the name `SPI2 `as it is defined by CMSIS.
More info are available here:
https://github.com/stm32duino/wiki/wiki/API#spi

fpistm, 2018-11-13:
Hi @Vicky-S 
Is this ok now?

Vicky-S, 2018-11-17:
Default SPI works well. But getting error in using 2nd SPI.

error: 'SPIClass' does not name a type

 SPIClass SPITwo(PB15, PB14, PB13);

 ^~~~~~~~

exit status 1
'SPIClass' does not name a type

Vicky-S, 2018-11-17:
Ok. Sorry my mistake. I didn't extern the SPI Library. It works fine.

Vicky-S, 2018-11-17:
Likewise how can I use several I2C and UART in STM32F103. ?

closed 2018-11-17 by Vicky-S

fpistm, 2018-11-17:
`HardwareSerial Serial2(USART2);`
or 
`HardwareSerial Serial2(PA3, PA2);`

Default I2C is on PB7/PB6 (D2/D3)
https://github.com/stm32duino/Arduino_Core_STM32/blob/646fff4c0703f32acf5db523d10687fbde4eca8b/variants/BLUEPILL_F103C8/variant.h#L104
So using Wire instance will use those pin.
else you can use the second I2C like that:
`TwoWire i2cTwo(PB11, PB10);`
or if you want change the default pin for Wire instance call this before `begin()`:
```
Wire.setSCL(PB10);
Wire.setSDA(PB11);
Wire.begin();
```
For uart/i2c/SPI, this is the same basic usage as Arduino except some API added which has been added for better flexibility. Example the setSDA/SCL. There is some other added for SPI (setMISO,...), serial (setRx,..)

Here I use stm32 pin name PXy but you can also use the pin number Dx or x.

Vicky-S, 2018-11-17:
Ok.  Thanks.