stm32duino/Arduino_Core_STM32 #371: Gaining access to CYCCNT from inside the setup() methode

hasenbanck, 2018-11-12:
In Marlin we need to have access to the DWT CYCCNT register for short delays. On a STM32F7 you need to execute following code to gain access to these register:

```c
CoreDebug->DEMCR |= CoreDebug_DEMCR_TRCENA_Msk;
DWT->LAR = 0xC5ACCE55;
DWT->CYCCNT = 0;
DWT->CTRL |= DWT_CTRL_CYCCNTENA_Msk;
```

This code is essential to not let the MCU halt when disconnecting the debugger probe (in my case Segger JLINK). After executing the above code, the program is then able to get the current cycle counter from the CYCCNT.

Right now we are calling the above code from the setup() methode as the first code. But sadly, this seems too late, since the MCU will halt when detaching the debugger probe and starting normally.

After some debugging I found out that I had to put the above code right after the SystemClock_Config() right after all clocks are configured. With that code change, I could now detach my debugger probe, the MCU didn't halt and the program got access to the CYCCNT register.

I need to look deeper into the issue and need to identify what between the SystemClock_Config() and the Setup() method is called, that results in this problem.

I would like to avoid putting the above code into the variant definition and I feel the setup() method is the right place to do so.

fpistm, 2018-11-12:
Hi @hasenbanck,
I've made a branch several months ago to perform some test around DWT.
Maybe this can help:
https://github.com/fpistm/Arduino_Core_STM32/tree/DWT

hasenbanck, 2018-11-13:
It seems you also had to setup the DWT access directly after the clock setup and before initializing the other peripherals. Maybe this is the correct way to do.

Suggestion:

We could add a function call inside the ```cores/arduino/stm32/hw_config.c```:

```c
void hw_config_init(void)
{
  //Initialize the HAL
  HAL_Init();

  // Configure the system clock
  SystemClock_Config();

  // Initialize optional debug peripheral
  SystemDebug_Config();
}
```

Whereas the ```SystemDebug_Config()``` would be a weak function, that could be overwritten by the sketch. So when, for whatever reason, the developer wants to configure the debugging peripherals (ITM/DWT or other Core Debug stuff), then he could use that entry-point to do so. 

@fpistm If you approve of such an idea, I would test such an implementation this evening and would open a PR.

fpistm, 2018-11-13:
Nils this make sense.
So agreed ;)

RickKimball, 2018-11-13:
Using the DWT stuff for timing doesn't seem like its best use. Why can't you use a real timer.   Doesn't the STM32F7 have more than enough?  

Regarding stuffing things into the main() function that you want to do early on in the c runtime setup, there are other ways to do this for your personal code.  I reject the idea that the core main() function needs to be modified to handle your personal code needs .  Check out the gcc specific attribute "constructor". You can use this attribute to force code to run before main().

https://stackoverflow.com/questions/2053029/how-exactly-does-attribute-constructor-work

NilsHasenbanck4758, 2018-11-13:
@RickKimball The delay function is not STM32F7 specific, so timers are not an option right now. The delay function are designed to be compatible for Cortex-M7 in general. So DWT is currently the best fit.

I will look if using the "constructor" attribute could help in this case.

hasenbanck, 2018-11-13:
Since the current premain() also uses the "constructor" attribute, we can indeed the suggested method by @RickKimball .

This is how we can do the initialization between the premain() and main() function of the core:
```
__attribute__(( constructor (102)))
void HAL_preinit() {
  CoreDebug->DEMCR |= CoreDebug_DEMCR_TRCENA_Msk;
  DWT->LAR = 0xC5ACCE55;
  DWT->CYCCNT = 0;
  DWT->CTRL |= DWT_CTRL_CYCCNTENA_Msk;
}
```

closed 2018-11-13 by hasenbanck

kolyshkin, 2019-02-21:
Just wondering where you guys are taking definitions of CoreDebug from?

I expected to find those in includes here, alas, no.

So I had to do this: https://github.com/FastLED/FastLED/pull/568/commits/c68b12ca003e0554f6

Perhaps there is a better way?

fpistm, 2019-02-21:
CoreDebug is a CMSIS  definition. For STM32F1, it is in the cortex m3 include file provided by the CMSIS package.