stm32duino/Arduino_Core_STM32 #373: Need help in improving SPI speed.

Vicky-S, 2018-11-18:
I have 38.4KSPS ADC(ADS1262) which have SPI interface. I'm trying to read the 38K samples in STM32F103C8t6. I need to send one byte command to get 4 bytes of data. So 5 bytes of data need to be transferred for each sample. Irrespective of SPI speed I can able to get max of 19K samples. I have tried to increase the SPI speed using clock divider function but the result is same.

For Single sample it took 45 micro seconds. but I need to get one sample with in 38us to get 38K samples. Initially my clock divider value was 16 then I increased it up to 2 but the time taken was same as 45us. How can it be?? Irrespective of clock speed the loop time remains same which is weird.

Is there a way to increase the speed of STM32F1? I can able to get 19k samples with 16MHz Arduino UNO but 72MHz STM32F1 gives the same output.

Vicky-S, 2018-11-18:
Yep. I tried but it takes more time than usual for me.

And im using only one device now so im keeping my CS pin to low for all the time.

fpistm, 2019-02-19:
Close this one to track all SPI request on the same issue #257.
Mark it as duplicated.

closed 2019-02-19 by fpistm