stm32duino/Arduino_Core_STM32 #374: Suggested additions to Arduino_Core_STM32

AndySymons, 2018-11-19:
Dear fpiSTM, 
As discussed on STM32duino, I am presenting two suggested additions; I am grateful for your offer to raise the PR(s). My private message to you on STM32duino gives more details. I attach a ZIP file with the referenced code. You are welcome to edit it as you see fit. I confirm that I claim no copyright to this code (though would appreciate a credit) and am happy for you to issue it under the usual terms. 
Kind regards,  
[$Andys boards.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/2597001/Andys.boards.zip)
Andy

fpistm, 2018-11-20:
Hi @AndySymons 
Thanks I will look at that and submit appropriate PR.
About credential, you will be the author of commit ;)

fpistm, 2019-01-07:
#376 merged

closed 2019-01-07 by fpistm