stm32duino/Arduino_Core_STM32 #375: Any news on CAN Bus - STM32F103

silardg, 2018-11-20:
I have been searching for a good supported CAN library for the STM32F103 (Bluepill).

I have found that there are supporting functions [here](https://github.com/stm32duino/Arduino_Core_STM32/tree/master/system/Drivers/STM32F1xx_HAL_Driver) for the CAN interface, but I'm not sure how to access them or use them.

silardg, 2018-11-20:
I have been digging into the files to understand it better.
I have found [this](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/BLUEPILL_F103C8/stm32f1xx_hal_conf.h#L55) define, where including the above referenced code is commented out. 

If I uncomment it, will I be able to access it somehow?

fpistm, 2018-11-20:
Hi @silardg 
Currently, it is not supported.
Anyway you can enable it and code it yourself using the HAL.
One other solution is to use Roger core which has a CAN library:
http://stm32duino.com/viewtopic.php?f=15&t=72

VictorLamoine, 2019-02-18:
Looks like CAN is still disabled for the Blue-Pill (https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/BLUEPILL_F103XX/stm32f1xx_hal_conf.h#L55)

Is there any plan to add CAN support for the Blue-Pill?

fpistm, 2019-02-18:
Currently no plan as I've a lot of stuff to do 😕 
I know @geosmall do some dev around CAN:
http://stm32duino.com/viewtopic.php?f=48&t=4524

fpistm, 2019-02-19:
Close this one to track all CAN request on the same issue #259.
Mark it as duplicated.

closed 2019-02-19 by fpistm