stm32duino/Arduino_Core_STM32 #383: Command-line builds not working

mithat, 2018-12-10:
I'm running Arduino 1.8.8 under Linux (Debian sid). I'm trying to configure a command-line build process for stm32duino projects using the `arduino` command line interface. It seems stm32duino isn't handling this properly. In particular:

### Specifying a build directory

When I issue a command to build into a specific directory:

```
/path/to/arduino --board stm32duino:STM32F1:genericSTM32F103C:cpu_speed=speed_72mhz,opt=osstd --pref build.path=release --verify build-test.ino
```

things seem to build as expected the first time.:

```
$ ls -1 release
build.options.json
build-test.ino.bin
build-test.ino.elf
build-test.ino.map
core
includes.cache
libraries
preproc
sketch
```

But on subsequent builds, thing go wonky. The compile fails:

```
Build options changed, rebuilding all
build-test:1:20: error: config.h: No such file or directory
 #include "config.h"
                    ^
compilation terminated.
exit status 1
```

and the contents of the build directory become:

```
$ ls -1 release
build.options.json
sketch
```

The file `build-test.ino` does indeed `#include config.h` file. The rebuild can't seem to find it.

### Not specifying a build directory

When I don't issue a command to build into a specific directory:

```
/path/to/arduino --board stm32duino:STM32F1:genericSTM32F103C:cpu_speed=speed_72mhz,opt=osstd --verify build-test.ino
```

things are built into a directory under `/tmp`, and when the build is complete, that directory vanishes.

Wiping of the `/tmp` directory is probably expected behavior with the CLI. (The IDE maintains the directory until you close it.) I mention it here only to point out that not specifying a build directory isn't a workaround.

mithat, 2018-12-10:
[build-test.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/2664740/build-test.zip)

fpistm, 2018-12-10:
Hi @mithat ,
 it seems you try to build this core: https://github.com/rogerclarkmelbourne/Arduino_STM32
Not this one.

mithat, 2018-12-10:
You're absolutely correct! Both projects refer users to [the same forum](http://www.stm32duino.com/), so I didn't realize they were different projects!

fpistm, 2018-12-10:
No worries. I close this one

closed 2018-12-10 by fpistm