stm32duino/Arduino_Core_STM32 #387: BLACK_F407VE button mappings incorrect?

BennehBoy, 2018-12-11:
Hi,

I'm having trouble using the buttons on a BLACK_F407VE.

The variant file suggests the following:

`// On-board user button
#define BTN_K_UP                PA0
#define BTN_K0                  PE4
#define BTN_K1                  PE5
#define USER_BTN                BTN_K0`

However, K1 appears to trigger via PE3, and K_UP does not trigger on PA0 (haven't isolated this yet).

Is it possible that this is being overridden elsewhere, or is it just plain wrong in the variant file?

Can anyone else confirm this?

fpistm, 2018-12-11:
It seems you're right:
http://wiki.stm32duino.com/images/5/5c/STM32_F4VE_SCHEMATIC.PDF
[image]

This should be:
```
#define BTN_K1                  PE3
```

fpistm, 2018-12-11:
K_UP is on PA0 but when it press it is pull up to 3.3V. So it is active HIGH. Configure it as INPUT_PULLDOWN.

BennehBoy, 2018-12-11:
Thought it may be something like that, thanks.

fpistm, 2018-12-12:
[image]