stm32duino/Arduino_Core_STM32 #393: BLACK_F407ZET6 support

uzi18, 2018-12-23:
Hi @fpistm 
I want to add this board as PR, do You think it is enough to copy BLACK_F407VE and modify a little?
I think big difference is number of IOs.
On the market it is possible also to buy BLACK_F407ZG module (1MB of flash).

fpistm, 2018-12-24:
Hi @uzi18 
yes this could be enough.
making a diff btw PeripheralPins.c only some pins added:
https://github.com/stm32duino/Arduino_Tools/blob/master/src/genpinmap/Arduino/STM32F407V(E-G)Tx/PeripheralPins.c
https://github.com/stm32duino/Arduino_Tools/blob/master/src/genpinmap/Arduino/STM32F407Z(E-G)Tx/PeripheralPins.c

If the pin mapping is nearly the same, the variant could be updated using board define as differentiator.
I've made it for L496 and L496-P. Same for ldscript.
https://github.com/stm32duino/Arduino_Core_STM32/blob/7b48eff35f425ab3d51106d4182bcd0e3cd30f74/variants/NUCLEO_L496ZG/PeripheralPins.c#L111

uzi18, 2018-12-24:
thx, will check this out

fpistm, 2018-12-27:
Yes, this is commented in the header:

>  * Note: Commented lines are alternative possibilities which are not used per default.
>  *       If you change them, you will have to know what you do

To sum up, only one pin config is available. If you do not comment duplicated pins only the first occurence is used so second is never used and  consume some space for nothing.

BennehBoy, 2019-01-05:
@fpistm I have 2 of these boards on order so started to look at building a variant.

Although PeripheralPins.c could easily be done with defines, the actual pin layout on the board is very different to the BLACK407VET6.  

The Variant.cpp file for BLACK407VET6 is laid out to represent the physical pin layout, accommodating the ZET6 here would necessitate either a) removing the physical layout altogether, or b) have a large ifdef section for each board.

Do you think it best to do one of the above, or break the ZET6 out into it's own unique variant folder?

uzi18, 2019-01-05:
docs are here: https://github.com/mcauser/BLACK_F407ZE

fpistm, 2019-01-05:
> Do you think it best to do one of the above, or break the ZET6 out into it's own unique variant folder?

There is no specific rules. If thanks a switch it's simple to support it then update the current variant.
Else if too much diff a new variant can be created.

uzi18, 2019-01-05:
@fpistm do you think it is possible to work with integrated sdcard slot, is sdio supported?

fpistm, 2019-01-05:
Yes SDIO is supported thanks STM32SD and FatFs library. Works on Black F407VE.

BennehBoy, 2019-01-05:
> Yes SDIO is supported thanks STM32SD and FatFs library. Works on Black F407VE.

I've tested this.

SPI works just fine on the 407VET6, so no reason to think it won't on this.

I think for the sake of simplicity I'll remove the 'physical layout of the 407VET6, that way one variant should suffice for VET6/ZET6/ZGT6.

uzi18, 2019-01-05:
It could be nice solution.
For fatfs did not find where to put files for STM32SD lib, it should be git submodule or something like that.

fpistm, 2019-01-05:
@uzi18 those are libraries avalable thanks arduino library manager.
https://github.com/stm32duino/FatFs
https://github.com/stm32duino/STM32SD

@BennehBoy we talk about SD via SDIO not SPI ;)

uzi18, 2019-01-05:
I need also SPI for cheap W5500 ethernet module or maybe there is better alternative?

BennehBoy, 2019-01-05:
> @BennehBoy we talk about SD via SDIO not SPI ;)

Yes, I've had both working.  SDIO using STM32SD, & SD via SPI using SdFat.

BennehBoy, 2019-01-05:
@uzi18 I've created a PR to add support if you are able to test.

uzi18, 2019-01-06:
```
Arduino_Core_STM32/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f407xx.h:183:81: fatal error: core_cm4.h: No such file or directory
 #include "core_cm4.h"             /* Cortex-M4 processor and core peripherals */
```
can't find core_cm4.h

BennehBoy, 2019-01-07:
What sketch are you using for the test?

fpistm, 2019-01-07:
@BennehBoy 
I've done a PR on your fork about Flash size management. Let me know what you think.

BennehBoy, 2019-01-07:
@fpistm seen and commented.

uzi18, 2019-01-08:
@fpistm You are right, it is impossible to build directly from git repo, so need to prepare fake core deploy dir, is there another way for test?

BennehBoy, 2019-01-08:
See here -> https://github.com/stm32duino/wiki/wiki/Using-git-repository

It explains how to use the git repo instead of the boards manager version (you must first have the boards manager version installed though).  I copy my old 1.4.0 folder off first so I can swap it back in if I want to just test against 1.4.0 release...  essentially when you clone this repo you will rename the folder to 1.4.0

Once you've done that, and have git tools installed, you can do the following:

`git config --global --add alias.pr '!f() { git fetch -fu ${2:-origin} refs/pull/$1/head:pr/$1 && git checkout pr/$1; }; f'`

Which means you can issue a command:
git pr 393

Which will download this pr, create a branch called 393, and switch to it.

uzi18, 2019-01-08:
have checked out pr 393 but didn't know that some other dirs are also needed
will just symlink other stuff from original

I think git submodules shoud do the job here

BennehBoy, 2019-01-08:
My board is on the slow boat from China so it would be good to see some test results :)

fpistm, 2019-01-08:
No gitsubmodule is not the right approach. The simpliest way is to clone the repo as described in the wiki then get the PR.

uzi18, 2019-01-08:
@fpistm my idea is to have STM32 repo with core,cmsis tools etc. as submodules inside, so you can fetch everything related with one command line
OK I'm trying to prepare now

fpistm, 2019-01-08:
Oh ok up to you ;)
There are several way to use git :)

uzi18, 2019-01-08:
building process works fine, now question, Serial object is mapped to Serial1 or SerialUSB?

fpistm, 2019-01-08:
By default Serial is mapped on Serial1 for this PR.
I will provide a way to map Serial on SerialUSB soon.
Anyway, it is possible to define it.
For example select no Serial in the menu then on top of your sketch `#define Serial SerialUSB` would work.
or define it in the variant.h thne HardwareSerail will not define it as already define or using build_opt.h.

uzi18, 2019-01-08:
ok I see, it is not possible to choose any options for USB cdc, when you plan to merge #388 ?

BennehBoy, 2019-01-08:
Simplest way would be to....

copy the relevant section from boards.txt in this pr
```
# Black F407Z(E-G)
# Support: Serial1 (USART1 on PA10, PA9)
# Default SPI: SPI (SPI1 on PB3,PB4,PB5 - NRF24 connector and W25Q16 (on board flash))
GenF4.menu.pnum.BLACK_F407ZE=Black F407Z(E-G)
GenF4.menu.pnum.BLACK_F407ZE.upload.maximum_data_size=131072
GenF4.menu.pnum.BLACK_F407ZE.build.mcu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard
GenF4.menu.pnum.BLACK_F407ZE.build.board=BLACK_F407ZE_G
GenF4.menu.pnum.BLACK_F407ZE.build.series=STM32F4xx
GenF4.menu.pnum.BLACK_F407ZE.build.product_line=STM32F407xx
GenF4.menu.pnum.BLACK_F407ZE.build.variant=BLACK_F407VE
GenF4.menu.pnum.BLACK_F407ZE.build.cmsis_lib_gcc=arm_cortexM4l_math

GenF4.menu.flash.ZE=ZE 512k (default)
GenF4.menu.flash.ZE.upload.maximum_size=524288
GenF4.menu.flash.ZE.build.ldscript=ldscript.ld
GenF4.menu.flash.ZG=ZG 1024k
GenF4.menu.flash.ZG.upload.maximum_size=1048576
GenF4.menu.flash.ZG.build.ldscript=ldscript_zg.ld
```

Make a copy of the black407VE variant folder

git pr-clean

git pr 388

copy the above saved black407VE folder over the top of the one that pr388 just updated

add the saved boards.txt items in (see the code snippet above).

This will manually merge my PR into #388 on your local working dir.

uzi18, 2019-01-08:
Can wait for usb support, just asking.
Need now to connect TTL converter to J5 onboard to see if test program work ok.

uzi18, 2019-01-08:
@BennehBoy ok for now simple blink on LED_BUILTIN does not work for me.
... Serial.begin/print also
what to check now? startup configuration?

BennehBoy, 2019-01-08:
Hmm, I would try referencing the LED pins directly rather than LED_BUILTIN

They should be on: PF_10  &  PF_9

Could be that the arduino numeric pin mappings for the pins are not correct - I couldn't find a schematic with them marked (such as this for VET6 -> http://wiki.stm32duino.com/images/3/3e/Black_STM32f4VET6_Pinouts_Left.pdf)

I actually used the Ver2 of those schematics which simply had the numeric numbering running consecutively as the pins progressed downwards - clearly incorrectly.

I'll do some digging - if you have a similar schematic for ZE then please share.

uzi18, 2019-01-08:
here you are: https://github.com/mcauser/BLACK_F407ZE/blob/master/docs/STM32F407ZET6_schematics.pdf

for me pin numbering looks ok (except D1/D2/D3 - D3 - is directly 3v3 powered), will check this one more time

uzi18, 2019-01-08:
@BennehBoy no change, now will check clock configuration etc..

BennehBoy, 2019-01-08:
D1 should start at PB12, ignore the top 4 pins which are power and ground.

BennehBoy, 2019-01-08:
Done, if you grab the PR again and restest that would be great.

uzi18, 2019-01-08:
@BennehBoy now is wrong with number of analogs ... 
 _Static_assert(NUM_ANALOG_INPUTS <= MAX_ANALOG_INPUTS,
 _Static_assert(NUM_ANALOG_FIRST >= NUM_ANALOG_INPUTS,
 _Static_assert(NUM_ANALOG_FIRST >= NUM_ANALOG_INPUTS,

BennehBoy, 2019-01-08:
It compiles fine for me.... where are you seeing that?

BennehBoy, 2019-01-08:
From Variant.h

```
// This must be a literal
#define NUM_DIGITAL_PINS        129
// This must be a literal with a value less than or equal to MAX_ANALOG_INPUTS
#define NUM_ANALOG_INPUTS       19
#define NUM_ANALOG_FIRST        110
```

uzi18, 2019-01-08:
ok after make clean, all build, but D2/D1 still does not blink checked with PF9 and PF_9 and PF10 and PF_10

BennehBoy, 2019-01-09:
I've looked at the STM32GENERIC definition and it has the LED's on the same pins.

The MicroPython config that you linked is contradictory, it has the LEDS on PA6 & PA7 in the pins.csv file, but PF9 & PF10 in the mpconfigboard.h

My board arrival eta is ~ 2 weeks away :/

uzi18, 2019-01-09:
@BennehBoy stm32generic works for me with these leds, tested months before.

BennehBoy, 2019-01-11:
Just an update to note that @uzi18 has confirmed that PR #402 does indeed work, the issue encountered was a local toolchain one.

fpistm, 2019-01-13:
Close this one as #402 merged

closed 2019-01-13 by fpistm

uzi18, 2019-01-13:
I still need to test I2C and timers.

fpistm, 2019-01-13:
yes

BennehBoy, 2019-01-13:
I'd be very surprised if i2c does not work, but please do test