stm32duino/Arduino_Core_STM32 #394: Hardware Flow control for serial communication

mayurharge, 2018-12-24:
Hello,
I am using the STM32f103c series controller with GSM module. Sometimes Serial buffer of GSM is getting full and the communication is not happening as expected. During my search, I came across Hardware Flow control using RTS/CTS. As stm has dedicated H/W flow control pins [For UART 2 PA0:- CTS2 & PA1:-RTS2] is there any option available in board files to enable hardware flow control? 
if yes please let me know how to enable it 
if not then it would be really helpful if you can integrate this in next version.

Thanks

fpistm, 2018-12-24:
HW flow control is not implemented.

mayurharge, 2018-12-24:
Okay, thanks.

mayurharge, 2018-12-25:
Thanks i will look into it

fpistm, 2018-12-26:
@mayurharge 
You could let this issue opened as a requested feature.
Do not hesitate to provide a PR if you get it up and running.

arihantdaga, 2018-12-26:
@fpistm  I saw this line in cores/arduino/stm32/uart.c https://github.com/stm32duino/Arduino_Core_STM32/blob/7b48eff35f425ab3d51106d4182bcd0e3cd30f74/cores/arduino/stm32/uart.c#L272 . Does making the HwFlowCtl to UART_HWCONTROL_CTS or UART_HWCONTROL_RTS is all thats required ? I am not sure yet, testing it. But if it works, I'll add this configuration using macros and make it user configurable...

fpistm, 2018-12-26:
I guess you have also to init the RTS/CTS pin also.

fpistm, 2018-12-26:
Just a quick thought if the buffer is full.
By default , the Rx and Tx buffer of the core is 64.
https://github.com/stm32duino/Arduino_Core_STM32/blob/bad5b56870ce311a4d591f789cbe5d47450097e4/cores/arduino/HardwareSerial.h#L42

You can redefined them by adding a new file (Tab on Arduino IDE) named **_build_opt.h_**
Then in this file put:
`-DSERIAL_RX_BUFFER_SIZE=256 -DSERIAL_TX_BUFFER_SIZE=256`

Increasing only one of both works also but I advise to keep them aligned.

mayurharge, 2018-12-26:
@fpistm I am already using the BUFFER_SIZE =256.
the need for me to use hardware flow control is that the slave device SIM800c is lagging behind over time 
When I searched for this issue  I figured out this

> Flow control is very important for correct communication between the GSM engine and DTE.
> For in the case such as a data or fax call, the sending device is transferring data faster than the
> receiving side is ready to accept. When the receiving buffer reaches its capacity, the receiving
> device should be capable to cause the sending device to pause until it catches up

This was mentioned in the datasheet of SIM800.

benwaffle, 2019-03-15:
@mayurharge you can of course use the HAL manually for now by calling HAL_GPIO_* and HAL_UART_* functions

fpistm, 2021-06-11:
Added to [U(S)ART] request list #1418

closed 2021-06-11 by fpistm

fpistm, 2022-02-05:
Done thanks #1634