stm32duino/Arduino_Core_STM32 #396: EEPROM Reset issue

mayurharge, 2018-12-25:
Hello,
I am using EMBEDIS library for handling EEPROM related functions.
I am getting the proper response for the first startup. but I am not able to retain the values after a restart.
the code works fine with [http://dan.drown.org/stm32duino/package_STM32duino_index.json]
board files.
Thanks

mayurharge, 2018-12-25:
Yes i have used controller from bluepill. stm32f103c series 128kb version.
Build options were default Just changed the upload method to serial and controller size to 128 kb.

Here is the link for sketch

https://github.com/thingSoC/embedis/blob/master/examples/Uno/Embedis_Uno/Embedis_Uno.ino

I will update the exact build options later

fpistm, 2018-12-28:
I will not be able to test before next week. Anyway, last time I've tested the eeprom emulation on F1 it was functionnal. I saw some core define usage in the lib maybe there is a small update to do also in the lib.
Did you test default eeprom example which do not use the lib? (Put and get example)

mayurharge, 2018-12-28:
Okay i wil test that.

Thanks & Regards,
   Mayur Harge <http://www.linkedin.com/in/mayurharge>
+919096028050
+918668423824

fpistm, 2019-01-03:
Hi @mayurharge 
I've been able to test and reproduce.
In fact value are not reset and well present in the FLASH.
Issue come from a wrong E2END definition. As EMBEDIS store since the end of the EEPROM it miss one data as E2END was defined as the SIZE instead of the last valid address.

mayurharge, 2019-01-03:
Can you please update where to define the valid E2END definition

Thanks & Regards,
   Mayur Harge <http://www.linkedin.com/in/mayurharge>

PR #400 (2019-01-04): [EEPROM] Wrong E2END definition
E2END is the last valid EEPROM address.
Fix #396
commit 93c5d894 (2019-01-04): [EEPROM] Wrong E2END definition

E2END is the last valid EEPROM address.
Fix #396
closed 2019-01-04 by fpistm