stm32duino/Arduino_Core_STM32 #398: How to have different instances of this core installed together?

edogaldo, 2018-12-27:
As per title: I'd like to have different instances of this core installed together (i.e. the main branch, the USB branch, etc.) so that I can switch each other as desired..

Thanks in advance and best, E.

edogaldo, 2018-12-27:
Thanks Fredric but I'm missing the answer to my question in the link you provided.
As I can understand the "Using git repository" page provides info on how to substitute version <x.y.z> with a different version (i.e. <a.b.c>).
What I'd instead need is to have both version <x.y.z> and <a.b.c> installed and available.

fpistm, 2018-12-27:
If you use the git you have all versions, simply by checkout branch or tag you want. 
Having several version is not possible with Arduino IDE or you have to manually change folder name or used several Arduino install with portable feature.

edogaldo, 2019-01-31:
Well, to me it would be better to allow having different folders with different core versions so that one could switch from one version to another just changing the board selection; this, by the way, would be consistent with the multi-core support of the Arduino ide.
Anyway, if it's too much complicated or not possible at all... ok! :)

Thanks again, E.

fpistm, 2019-01-31:
@edogaldo 
This relies from Arduino IDE not from the core.
I think (maybe) you can have 2 install one in hardware folder of the Arduino IDE and the one in package/  folder but this also relies to Arduino IDE.
I guess you should ask this on Arduino forum or open an issue on Arduino IDE to ask if possible to have several core installed. Maybe an issue is already open for that.

edogaldo, 2019-01-31:
Ok, I addressed this as a core issue since this is possible with Roger's core (for example) without any conflicts..

BennehBoy, 2019-01-31:
I think it's not really a use case for most people - anyone who needs to do this is on the development bleeding edge and can self support via github checkouts/hand merging, having multipl core folders that they swap in/out of the current core folder location.

AnHardt, 2019-01-31:
Inconsistency.
A bit further down i read
```
So, you should have two directories in \<Arduino install directory\>/hardware/stm/
stm32 and tools
```
That worked for me.

AnHardt, 2019-01-31:
Tried renaming 
"C:\Program Files (x86)\Arduino\hardware\stm"
to 
"C:\Program Files (x86)\Arduino\hardware\STM32".
Arduino IDE does not come up anymore.

fpistm, 2019-01-31:
Well, I could not tell why, probably cache data issue in arduino index. Stay with stm if it works. All those stuff is Arduino IDE related. I provide a guide to help but end user have to know what and why he do this and how Arduino handle all cores. As mention in the Readme.md this is for advanced user.
Roger core is not referenced thanks the board manager so this is easiest to manage.

uzi18, 2019-01-31:
I use master branch of this core on Sloeber/eclipse, but first did installed 1.4.0 and linked this repo as 1.9.9 ;)
You can choose what you want there,
So You are able to have couple of versions but need to create different folders for them.

edogaldo, 2019-01-31:
Will perform further experiments.
Thanks all for suggestions.

E.

closed 2019-01-31 by fpistm