stm32duino/Arduino_Core_STM32 #403: AnalogReference

Yannick-Marietti, 2019-01-07:
For compatibility concerns, is it possible to implement a "dummy" analogReference() function ?

This function is often used by teachers to explain ADC principle using Arduino UNO boards.
Of course, if STM32  ADC can be set to a level different than 3.3 Volts, a future working implementation would be a must-have for teachers.

BR

Yannick

closed 2019-06-05 by fpistm

firexx, 2023-04-11:
in the current version 2.5.0 used 
`void analogReference(eAnalogReference ulMode) `
prototype, instead of original in Arduino:
` void analogReference(uint8_t mode);`
 so the compatibility is not provided

fpistm, 2023-04-11:
Original was with `eAnalogReference` which is an enum.
Example in the samd core:
https://github.com/arduino/ArduinoCore-sam/blob/790ff2c852bf159787a9966bddee4d9f55352d15/cores/arduino/wiring_analog.h#L40

Now within the Arduino Core API: 
https://github.com/arduino/ArduinoCore-API/blob/master/api/Common.h#L100

anyway, the mode always defined as an enum but with different name like with the mbed core:
https://github.com/arduino/ArduinoCore-mbed/blob/0504d34ada3d5aac69eacb481bcdc64860aebca9/variants/NICLA/pins_arduino.h#L13

As the function is provided just for compatibility (be able to build) and do nothing as the arguments is unused and also match an `uint8_t `value it is compatible.

PR #535 (2019-06-05): Add dummy analogReference() for compatibility
Reference:
https://www.arduino.cc/reference/en/language/functions/analog-io/analogreference/

Same as Arduino SAM Boards (Due)

Fix #403
commit dbf538e1 (2019-06-05): Add dummy analogReference() for compatibility

Reference:
https://www.arduino.cc/reference/en/language/functions/analog-io/analogreference/

Same as Arduino SAM Boards (Due)

Fix #403