stm32duino/Arduino_Core_STM32 #407: New variant: Naze32 rev5

MauroMombelli, 2019-01-13:
It is exactly like "Generic STM32F103 series" but HDE clock is 12.000.000.

Right now i copy pasted the generic folder, set HDE define, and added in boards.txt
```
# NAZE32_REV5 board
# Support: Serial2 (USART2 on PA3, PA2) and Serial3 (USART3 on PB11, PB10)
GenF103.menu.pnum.NAZE32_REV5=NAZE32_REV5
GenF103.menu.pnum.NAZE32_REV5.upload.maximum_data_size=20480
GenF103.menu.pnum.NAZE32_REV5.build.mcu=cortex-m3
GenF103.menu.pnum.NAZE32_REV5.build.board=NAZE32_REV5
GenF103.menu.pnum.NAZE32_REV5.build.series=STM32F1xx
GenF103.menu.pnum.NAZE32_REV5.build.product_line=STM32F103xB
GenF103.menu.pnum.NAZE32_REV5.build.variant=NAZE32_REV5
GenF103.menu.pnum.NAZE32_REV5.build.cmsis_lib_gcc=arm_cortexM3l_math
```

This ass a sub-variant; i wonder if there is a way to specity the define from the boards.txt only.
I also have a little example that will print on the serial the onboard MPU6050 sensor, and provide the mapping for input/output pin in the board (it is closed hardware, so is a bit hard to find out)

BennehBoy, 2019-01-13:
Perhaps a flight controller section might be appropriate, there are a lot of them out there - given how small the variant files are I don't think it matters too much about the duplication.

Of course @fpistm  might have other ideas.

fpistm, 2019-01-14:
You could add a line in boards.txt to override extra_flags:
`GenF103.menu.pnum.NAZE32_REV5.build.extra_flags=-D{build.product_line} {build.enable_usb} {build.xSerial} -DHSE_VALUE=12000000`

Adding a new variant could be fine if the pin mapping needs customization.
Mainly, if pins are dedicated to one function then you can optimize `PeripheralPins.c` to only have those available and save space.

fpistm, 2019-01-27:
You have to restart Arduino IDE to take in account board.txt changes.

fpistm, 2019-01-27:
Right `build.variant` is mandatory. 
For `GenF103.menu.pnum.NAZE32_REV5.build.board=BLUEPILL_F103C8`
You can set your board name:
`GenF103.menu.pnum.NAZE32_REV5.build.board=NAZE32_REV5`

MauroMombelli, 2019-01-28:
yes, i know, the problem was that you must provide a build.variant (I think), so after changing random stuff, this seems to work:

```
GenF103.menu.pnum.NAZE32_REV5=NAZE32_REV5
GenF103.menu.pnum.NAZE32_REV5.build.extra_flags=-D{build.product_line} {build.enable_usb} {build.xSerial} -DHSE_VALUE=12000000
GenF103.menu.pnum.NAZE32_REV5.build.mcu=cortex-m3
GenF103.menu.pnum.NAZE32_REV5.build.series=STM32F1xx
GenF103.menu.pnum.NAZE32_REV5.build.product_line=STM32F103xB
GenF103.menu.pnum.NAZE32_REV5.build.cmsis_lib_gcc=arm_cortexM3l_math
GenF103.menu.pnum.NAZE32_REV5.build.board=BLUEPILL_F103C8
GenF103.menu.pnum.NAZE32_REV5.build.variant=BLUEPILL_F103C8
```

fpistm, 2019-03-26:
Any update on this?
Did you plan to submit a PR?

fpistm, 2019-04-03:
Close as no feedback. 
OP can submit a PR for review if needed.

closed 2019-04-03 by fpistm