stm32duino/Arduino_Core_STM32 #413: enhancement request: add support for HY-TinySTM103T board

mitchmitchell, 2019-01-21:
Please add support for the HY-TinySTM103T boards as discussed here:  http://www.stm32duino.com/viewtopic.php?f=31&t=853

BennehBoy, 2019-01-25:
PR created, see -> #416

BennehBoy, 2019-02-09:
@mitchmitchell this still needs testing BTW

mitchmitchell, 2019-02-09:
Let me see if I can get Mysensors to compile with it and put it on one of my test boards.

mitchmitchell, 2019-02-09:
Not having much luck getting the board files to work -- is there some trick to picking up unpublished board files?  I try cloning the repo into the hardware directory and that didn't work out.

BennehBoy, 2019-02-10:
https://github.com/stm32duino/wiki/wiki/Using-git-repository

mitchmitchell, 2019-02-11:
Well it looks like the pin numbers aren't right:

Roger Clark's Arduino_STM32 pin numbers:
//
// Start oneWireSearch.ino
//
PA0 = 0
PA1 = 1
PA2 = 2
PA3 = 3
PA4 = 4
PA5 = 5
PA6 = 6
PA7 = 7
PB0 = 16
PB1 = 17
PB2 = 18
PB3 = 19
PB4 = 20
PB5 = 21
PB6 = 22
PB7 = 23

findDevices
findDevices
findDevices
findDevices
uint8_t pin22[][8] = {
  {
0x28, 0xFF, 0x7A, 0xCB, 0x02, 0x17, 0x04, 0x99  },
  {
0x28, 0xFF, 0x81, 0x3D, 0x02, 0x17, 0x05, 0x69  },
};
// nr devices found: 2

//
// End oneWireSearch.ino 
//

STM32Duino output:
//
//
// Start oneWireSearch.ino
//
PA0 = 2
PA1 = 1
PA2 = 0
PA3 = 11
PA4 = 12
PA5 = 13
PA6 = 14
PA7 = 15
PB0 = 16
PB1 = 17
PB2 = 18
PB3 = 9
PB4 = 8
PB5 = 7
PB6 = 6
PB7 = 5

//
// End oneWireSearch.ino 
//

Here is the test sketch:

//
//    FILE: oneWireSearch.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.02
// PURPOSE: scan for 1-Wire devices + code snippet generator
//    DATE: 2015-june-30
//     URL: http://forum.arduino.cc/index.php?topic=333923
//
// inspired by http://www.hacktronics.com/Tutorials/arduino-1-wire-address-finder.html
//
// Released to the public domain
//
// 0.1.00 initial version
// 0.1.01 first published version
// 0.1.02 small output changes

#include <OneWire.h>

void setup()
{
  Serial.begin(115200);
  Serial.println("//\n// Start oneWireSearch.ino\n//");
  Serial.print("PA0 = ");
  Serial.println(PA0);
  Serial.print("PA1 = ");
  Serial.println(PA1);
  Serial.print("PA2 = ");
  Serial.println(PA2);
  Serial.print("PA3 = ");
  Serial.println(PA3);
  Serial.print("PA4 = ");
  Serial.println(PA4);
  Serial.print("PA5 = ");
  Serial.println(PA5);
  Serial.print("PA6 = ");
  Serial.println(PA6);
  Serial.print("PA7 = ");
  Serial.println(PA7);
  Serial.print("PB0 = ");
  Serial.println(PB0);
  Serial.print("PB1 = ");
  Serial.println(PB1);
  Serial.print("PB2 = ");
  Serial.println(PB2);
  Serial.print("PB3 = ");
  Serial.println(PB3);
  Serial.print("PB4 = ");
  Serial.println(PB4);
  Serial.print("PB5 = ");
  Serial.println(PB5);
  Serial.print("PB6 = ");
  Serial.println(PB6);
  Serial.print("PB7 = ");
  Serial.println(PB7);

  for (uint8_t pin = PB3; pin < PB7; pin++)
  {
    findDevices(pin);
  }
  Serial.println("\n//\n// End oneWireSearch.ino \n//");
}

void loop()
{
}

uint8_t findDevices(int pin)
{
  Serial.print("\nfindDevices");
  OneWire ow(pin);

  uint8_t address[8];
  uint8_t count = 0;

  if (ow.search(address))
  {
    Serial.print("\nuint8_t pin");
    Serial.print(pin, DEC);
    Serial.println("[][8] = {");
    do {
      count++;
      Serial.println("  {");
      for (uint8_t i = 0; i < 8; i++)
      {
        Serial.print("0x");
        if (address[i] < 0x10) Serial.print("0");
        Serial.print(address[i], HEX);
        if (i < 7) Serial.print(", ");
      }
      Serial.println("  },");
    } while (ow.search(address));

    Serial.println("};");
    Serial.print("// nr devices found: ");
    Serial.println(count);
  }

  return count;
}

also, #define LED_BUILTIN PA1 not PB1 as on the maple

mitchmitchell, 2019-02-11:
#define LED_BUILTIN             PB1 should be PA1 in variants\HY_TinySTM103T/variant.h

mitchmitchell, 2019-02-11:
I suspected this might be the case given the different structure of the variant files and such.  I can always create an array of pins to loop over.  Am I correct in thinking that the pin names will match what is on the silk screen and schematic?  E.g. PB6 in the software means PB6 on the physical board as defined by the schematic/silk screening on that board?

fpistm, 2019-02-11:
Of course PB6 is PB6.

mitchmitchell, 2019-02-11:
Cool, I just wanted to not assume anything -- so BennehBoy asked me to test the varient -- it looks like NodeManager/MySensors HAL layer will need to be modified before it can run on the stm32duino definitions instead of Roger's Arduino_STM32 so I will need to create some test sketches -- any thoughts on areas you'd like to see covered with test cases?